luc-github/ESP3D-WEBUI #2: For the Documentation

leseaw, 2017-05-20:
My 8266 had to be completely overwritten with new firmware, otherwise would not have been visible after uploading the new GUI. Maybe write in the documentation?

luc-github, 2017-05-20:
It is mentioned in Installation part:
https://github.com/luc-github/ESP3D-WEBUI#installation

Installation

Please use the latest [ESP3D firmware](https://github.com/luc-github/ESP3D/tree/devt) and copy the index.html.gz file on root of SPIFFS, in theory ESP3D have a version of web-ui but it may not be the latest one

leseaw, 2017-05-20:
I had to flash my firmware for the board completely.
This i mean, not your firmware.

luc-github, 2017-05-20:
sorry I misunderstood which board ? which Firmware ?

leseaw, 2017-05-20:
NodeMCU V1 LoLin V3
https://nodemcu-build.com/
Build this Firmware.
From V0.62 to V0.9 i need to flash my firmware new.
Otherwise, despite successful uploading the web server would not have been a Wifi connection.

luc-github, 2017-05-20:
Sorry I am not aware of this, but with this lua FW as I guess it is Lua FW
the interface won't work - 

ESP3D-webui is for ESP3DD firmware not nodemcu FW

I am not sur to understand what issue you raise

leseaw, 2017-05-23:
close this :)
I will write an extra step.

closed 2017-05-23 by leseaw