luc-github/ESP3D-WEBUI #4: Feel free to feedback

luc-github, 2017-07-16:
Every constructive comment is welcome 😸

onekk, 2018-05-31:
First thing many thanks for this project (FW for the ESP8266 and this web UI).

It is very good and seem running well, but the UI is too big for my netbook.

It will be possible to add a smaller UI for smaller screen.

The UI is a good thing, but for example the position are out of the screen, it will be better to place them immediately under the movement button, and before the macro buttons, so when the movement are done it will be not necessary to scroll the UI to view the position (XY & Z).

One other thing, the console is at the bottom, maybe it will work better if it was swapped with the SD content.

ASA I will make a screenshot of my netbook and I will report the dimension in pixel.

I suppose that a more compact UI will be appreciated also by the folks with a Tablet as a Controller for the Printer.

luc-github, 2018-05-31:
Sure screenshot and resolution are always good to understand issues

Be noted ESP3D 2.0 has more responsive UI against resolution : https://github.com/luc-github/ESP3D-WEBUI/tree/asyncUI

onekk, 2018-05-31:
Ok it is usable or it is too rough for everyday use.

My printer uses Marlin and my ESP is an ESP12E (cheapest china clone at 1 Euro).

regards

Carlo D.

luc-github, 2018-05-31:
Compact UI is something that was raised before and need to be thinked properly - I just have lack of time to sit, think and design a complete compact UI - once UI would be defined should not be too long to code it
I should confess I am not UI professional and have a lack of knowledge of what resolutions need to be tested - no mention that depending browser UI change / control size changes / etc...

I agree would be useful but I am not sure currently how to do it right

luc-github, 2018-12-09:
Latest UI has improved support for low res device
as telnet support allow remote - some Android App works using it 
So far only file upload is not working for what I tested

As no update for month on this topic I close it

closed 2018-12-09 by luc-github