luc-github/ESP3D-WEBUI #5: Add check box for autocheck print status

luc-github, 2017-08-20:
the check box will act the 2 others allowing a periodical check of % progression using M27 command result

luc-github, 2017-09-13:
M27 format change depending of target FW Marlin give % when repetier give octet read from SD vs total size of GCODE file and not sure it works for host print status

cemof1, 2019-03-26:
hello luc
I'm using marlin and adding this feature will be great for me. I can't see the print status right now. Are you thinking of sharing additions?

luc-github, 2019-03-26:
Yes it will be in 3.X - M27 is the command you can use  in macro until it is implemented

heitorado, 2019-12-06:
Hi luc, really like your work with the ESP3D. I'm currently using it and helped me a lot so far!

I would like to help with this feature, if you want to. It would be really neat to have a progress bar on the web UI and I was thinking about also making it send a browser notification. I usually spend a lot of time working on the computer while the printer works and I think it would be nice to be notified when a print finishes. What do you think about it?

If you would like help, just tell me what I can do and how to start :)

luc-github, 2019-12-06:
for end print notification you can use the email / line /pushover notification https://github.com/luc-github/ESP3D/wiki/Notifications

The issue is this PR need to be tested on all FW (Smoothie/Marlin (2 and 1.1.9) / Marlin Kimbra / Repetier/ Marlin_esp32 / GRBL_ESP32 as each fw use different code to get progress and give different output 

Also this need to be tested on several resolutions (PC/Phone/Tablet) 

I did not have time to setup all system and test it yet - 

be notifications do no need to have webUI  open and poiling the status, which is more flexible IMHO

luc-github, 2020-09-10:
in 3.0 the print status is done by polling commands which does nort need any check box anymore
[image]

Display is done in notification area 

[image]

In marlin flavors file name is also be displayed if printing is detected

closed 2020-09-10 by luc-github