luc-github/ESP3D-WEBUI #7: Add Translation for status page

luc-github, 2017-09-06:
Status page is not translated in FR/SP/GE
[image]

3d-gussner, 2017-11-03:
This is on that is not translated. Also the "Choose file" button or "No file chosen" aren't translated.

luc-github, 2017-11-03:
The button and edit for upload is a system control- so it use local language I did not overrided it - may be would be better

3d-gussner, 2017-11-04:
As a German i still use my Windows in English (with German keyboard settings), because I am an IT professional and had always troubleshoot non-english Microsoft Workstations, Servers and apps.
But sometimes it is nice to read things in German.
For me the translation to other languages is to make these things also available to non-englsih speaking people. Especially children should be able to use things in their language.

luc-github, 2017-11-04:
I agree - that is why the UI propose translation - but for the upload control just need to spend time to override it - there toons of example on internet - I just did not spend time on it yet -

luc-github, 2017-12-11:
Close as duplicate as https://github.com/luc-github/ESP3D-WEBUI/issues/8
I have created a ticket for the upload button not translated

closed 2017-12-11 by luc-github