luc-github/ESP3D-WEBUI #8: Some translation are incompleted

luc-github, 2017-09-06:
Due to new setup wizard and settings page translation and coming status page - some German translations are missing
Help is more than welcome

3d-gussner, 2017-11-03:
Hi Luc,

just made a pull request. There are some sub tabs i could not translate. At this moment i also have issues to use node.js and gulp to test it. I am using a Win 10 64bit.
I could also translate to Dutch if needed.

luc-github, 2017-11-03:
Thank you 
What issue do you meet ? - I use n on Win7 and Win10 both 64b without issue
1- install nodejs
2 - go to ESP3D directory using terminal (cmd.exe)
3 - launch every command present in install.bat
```
npm install --global gulp-cli
npm install --save-dev gulp
npm install --save-dev gulp-jshint
npm install --save-dev jshint
npm install --save-dev gulp-if
npm install --save-dev gulp-concat
npm install --save-dev gulp-uglify
npm install --save-dev gulp-clean-css
npm install --save-dev gulp-remove-code
npm install --save-dev del
npm install --save-dev gulp-zip
npm install --save-dev gulp-gzip
npm install --save-dev gulp-htmlmin
npm install --save-dev gulp-replace
npm install --save-dev gulp-smoosher
```
4 - type `gulp package` to build index.html.gz with your changes

It should work

Sure every translation is welcome- thank you

onekk, 2018-05-31:
Hi, I'm managing to translate the interface in Italian.

How to test the translation?

I have cloned the repo usign git clone to muy local devel and now I have a **it.js** in **language** subdirectory.

Here the file
[trans-it.txt](https://github.com/luc-github/ESP3D-WEBUI/files/2059038/trans-it.txt)

Regards

Carlo D.

luc-github, 2018-05-31:
Salve, ^_^
Thank you - excellent !
I just integrated your file - I have modified some part to use HTML char instead of accents
https://github.com/luc-github/ESP3D-WEBUI/commit/d6dfc62490c7f5b88143a2536f23c8f49e2210ea

You can review it

if you have time can you check the 2.x for Italian ? - I enclose spanish version for the complete list of words to translate

[es.zip](https://github.com/luc-github/ESP3D-WEBUI/files/2059111/es.zip)

onekk, 2018-06-01:
Hi Luc, I have managed to translate the V 2.0 file, as a base I have used a diff from the es.js of the v1 and the es.js you kindly send me.

So I'm not 100% sure that it is correct, plus As I0ve not a V2 installed, I have guessed some translation, and some I've left untraslated from the English text.

So here my work, ASA you integrate it in the UI file i promise i will download it and check the V2 interface on my printer.

As I've a ATMEGA 2560 controller running MK4duo (Marlinkimbra) I hope I could test almost all the string, but feel free to contact me if something is wrong.

Best Regards.

Carlo D.
[it-v2.0.txt](https://github.com/luc-github/ESP3D-WEBUI/files/2061683/it-v2.0.txt)

luc-github, 2018-06-01:
Hi Carlo, thanks a lot ^_^, I will integrate today

onekk, 2018-06-01:
Hi, Luc, I have checked a little with my configuration and found that some error are left in the files, so after some other work, i have managed to make two new files, (hope they are more correct).

Some word have changed due to a better learning of the contest in which them are in the interface, some other are left untranslated, as in Italian is a common practice to leave some term untranslated,  i.e. Files and Directory but also Router and Access Point and Baud Rate.

For the interface I have to check all the buttons and the screens, but this work is left for another time, after the Week End.

Here two files V1.0 for the standard interface and V2.0 for the new interface.
for a better handling I have added the date of the translation.

Sorry for the double work in so little time, but if you haven't implemented the italian translation I have not been able to check the files in the  ESP3D on my printer.

[it-v20-20180601.txt](https://github.com/luc-github/ESP3D-WEBUI/files/2063265/it-v20-20180601.txt)
[it-v10-20180601.txt](https://github.com/luc-github/ESP3D-WEBUI/files/2063266/it-v10-20180601.txt)

luc-github, 2018-06-01:
Thank you I will update repo later ^_^

luc-github, 2018-10-03:
GRBL_ESP32 need a lot of translation in 2.1 - will prepare it soon

luc-github, 2018-12-09:
WiKi has now the translations status and the template/files 
I am closing this topic then

luc-github, 2019-05-20:
I reopen it as some translations were not included in language.tpl and some language are still incomplete 
https://github.com/luc-github/ESP3D-WEBUI/wiki/Translation-support

kondorzs, 2020-06-10:
HI LUC, I transolated index.html and itegrated HUNGARIAN language.  I tested work good.

How can I send it to you?

luc-github, 2020-06-10:
@kondorzs thank you but can you use the template file https://raw.githubusercontent.com/luc-github/ESP3D-WEBUI/2.1/www/js/language/lang.tpl like explain in wiki : https://github.com/luc-github/ESP3D-WEBUI/wiki/Translation-support

using the index.html is too fuzzy

kondorzs, 2020-06-10:
ok I will put Hungarian language in template file I will cut from index.html.

luc-github, 2020-06-11:
@onekk 
Sorry I missed your question :
> What is the proper standard? unicode characters are accepted as is at least for the accented vovels, or the html form is better?
In 2.1 the html form is better for special char, for some reason the conversion is not UTF8, so display is garbage for browser not having these chars table - 
 in 3.0 it is fully UTF8 and it won't be necessary anymore

luc-github, 2020-06-11:
integration done https://github.com/luc-github/ESP3D-WEBUI/commit/069771a3f34117054bcfb5cb26a8e6c350a2dc23 
Thank you ^_^

kondorzs, 2020-06-11:
I thank you, please.

Br
Zsolt

kondorzs, 2020-06-11:
Hi, I tested new firmware and languages files, work very vell, but I missed one word transolated no visible. Reason is one space. 

bad
"**TargetFW**":"Cél FW",

good

"Target FW":"Cél FW",

[image]

correct hu.js is:

[hu.zip](https://github.com/luc-github/ESP3D-WEBUI/files/4763542/hu.zip)

I'm sorry....

luc-github, 2020-06-11:
Thank you, fixed with some others typo I have noticed https://github.com/luc-github/ESP3D-WEBUI/commit/f137b71e33e253bcbd33c26eab396756bc10931d

I have noticed tpl is not up to date and still miss some translations (e.g for Notifications) but will fix that later

kondorzs, 2020-06-11:
OK Thanks

luc-github, 2023-10-03:
Closing for 2.1, will create new one for 3.0 when ready

closed 2023-10-03 by luc-github