luc-github/ESP3D-WEBUI #19: Add Toshiba flash air support for file upload

luc-github, 2018-01-09:
As Serial upload is slow and it exists wifi SD card like toshiba flashair - so would be nice to use the flashair feature for the upload and even the SD card free space availability 

[image]

Nappen, 2018-04-22:
Has it been any updates on this? because this is my exact question as well. i have flashair on my printers. and want to run it with the esp3d

luc-github, 2018-04-23:
It is in TODO list.  Now there is a preferences feature it is easier to configure the UI and so to add this feature.
Waiting for this you can use camera panel if you do not use camera

luc-github, 2018-04-23:
but the implementation won't be like camera = an iframe, the target is to make the flashair to generate a json file and use existing UI not replace existing file panel by an iframe panel linked to flashair index.html

this will keep the ui consistant

Nappen, 2018-04-23:
Ahh would work. U can run a custom website on the flashair so it will look the same as the interface of esp3d. Right now i run the extrude3d guys interface on the sd card. Try it today with the camera thanks for the quick respond

luc-github, 2018-04-23:
Yes I know I did one already https://github.com/luc-github/Flashair-FM

luc-github, 2018-12-09:
The implement will be not as smooth as expected as toshiba flash air does not support cross origin - so iframe is the only solution as far I know

luc-github, 2020-10-29:
with external panel support now WiFi SD can be  displayed and used to upload
[image]

The current Flash air FM should be adapted to better fit the panel UI but it currently show that even not optimized any UI is usable

luc-github, 2020-10-30:
I guess this one can be closed - new UI need to be done in other repository

closed 2020-10-30 by luc-github