luc-github/ESP3D-WEBUI #21: More extruder

kina500, 2018-03-15:
Hi

Thank you very much for perfect wifi and webui for 3D printer :-)
I am using 6 extruders and one hotend on my printer (with a filament splitter).

How do I add more extruder to ESP3D-WEBUI?

luc-github, 2018-03-15:
It is mixed extruder ?
what command do you need ? I am not familiar with mixed extruder
I guess 1 temperature control but 6 positions control ?

kina500, 2018-03-15:
Yes, you are absolutely right :-)
1 temp control and 6 position for extruder

luc-github, 2018-03-15:
how about to add a drob box instead of T0 to select extruder:
[image]
Because adding a line per extruder like for a second extruder will be hard to fit panel

Also can you share what result the M105 command give ? to see if mixing extruder temperature output is affected

kina500, 2018-03-15:
A dropbox would be fine :-)
Maybe call them E0 to Ex instead, as the temp in hotted is T0.
Only if you have same amount of hotends as extruders, then T0 is both hotted and extruder.

If i put M105 in, I get:
M105
T:21.83 /0 B:23.06 /0 B@:0 @:0

luc-github, 2018-03-15:
Ok Thanks - seems no impact on temperatures - which is good 
I will add mixed extruder feature in preferences including number of extruders

luc-github, 2018-03-16:
Until this is implemented you can use the macro button for E1->E5 as E0 is already on UI

luc-github, 2018-06-17:
Implemented upstream - will be pushed soon

luc-github, 2018-07-17:
pushed already

closed 2018-07-17 by luc-github