luc-github/ESP3D-WEBUI #22: Macro disappair

kina500, 2018-03-15:
Hi

Tried to make macro, as this is a very nice feature.

Just for test, I made a text file containing G28 X0 and call it HomeX.G.
When I save the file on ESP, and execute the macro, it expect to be a command for the ESP device and give an error.
Is that intension?
I saved the Gcode file on SD instead, and it works perfect.
After I closed the web browser and went back into ESP3D WEBUI, the macro is gone...
Do I have to edit the macrocfg.json file on the ESP device so that the device remember the macro's?

kina500, 2018-03-15:
Tried to edit the macrocode.json, but it still forget the macro.
I guess I did put wrong data in it?

[
 {
  "name": "Home X",
  "glyph": "1",
  "filename": "HomeX.G",
  "target": "SD",
  "class": "",
  "index": 0
 },
 {
  "name": "",
  "glyph": "",
  "filename": "",
  "target": "",
  "class": "",
  "index": 1
 },
 {

luc-github, 2018-03-15:
Having error is never the intention ^_^
Macro code have been updated recently due to some bug - do you use the latest git version ?
it is better not to update the macro.json manually use the macro button which set proper values, if file is corrupted it will use default empty value.

kina500, 2018-03-15:
I have to download master branch, is that the latest?
How do I update, just upload all new files? index and macrocfg?

luc-github, 2018-03-15:
the macro fix is in ESP3D FW  https://github.com/luc-github/ESP3D, you need to reflash the master git

To get latest web UI you just need to update the index.html.gz from here : https://github.com/luc-github/ESP3D-WEBUI/blob/master/index.html.gz

kina500, 2018-03-15:
Ok, thank you very much :-)

kina500, 2018-03-16:
Cannot upload newest index.html.gz, then I cannon connect anymore.
When I upload it through the WEBUI, it only show filesize of 2Kb

kina500, 2018-03-16:
Everything went smooth with 0.99 version.
Downloaded firmware from master. 
Downloaded the newest index.html.gz from your link.
Exchanged the file in data folder of ESP3D-Master folder.
Flashed the firmware, went ok.
Used the 8266 sketch uploader tool in Arduino IDE, and got error message:

Arduino: 1.8.5 (Windows 7), TD: 1.41, Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Archiving built core (caching) in: C:\Users\f09721\AppData\Local\Temp\arduino_cache_862136\core\core_esp8266_esp8266_generic_CpuFrequency_80,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_qio,FlashSize_512K0,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_all,UploadSpeed_115200_84b2d07d828a7ecb3156ad4b70b75b0a.a
Sketch uses 373144 bytes (74%) of program storage space. Maximum is 499696 bytes.
Global variables use 37548 bytes (45%) of dynamic memory, leaving 44372 bytes for local variables. Maximum is 81920 bytes.
Erasing 0x80000 bytes starting at 0x00000000
Uploading 377296 bytes from C:\Users\f09721\AppData\Local\Temp\arduino_build_360452/esp3d.ino.bin to flash at 0x00000000
................................................................................ [ 21% ]
................................................................................ [ 43% ]
................................................................................ [ 65% ]
................................................................................ [ 86% ]
.................................................                                [ 100% ]

SPIFFS Not Defined for Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200

kina500, 2018-03-16:
Just tried again to verify if I did something wrong.
First I flashed 0.9.99 again and data with 8266 sketch uploader tool in Arduino IDE.
Everything went well

Then I tried with ESP3D-MASTER version, with following error (the index.html.gz was not exchanged this time):

Arduino: 1.8.5 (Mac OS X), TD: 1.41, Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 1M (512K SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Archiving built core (caching) in: /var/folders/sm/tgqgjcwx3fd3zj4_fh5nq4tw0000gn/T/arduino_cache_413132/core/core_esp8266_esp8266_generic_CpuFrequency_80,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_qio,FlashSize_1M512,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_all,UploadSpeed_115200_ae02655205d10a8fb66bd35d4a4f489c.a
Sketch uses 371376 bytes (74%) of program storage space. Maximum is 499696 bytes.
Global variables use 36940 bytes (45%) of dynamic memory, leaving 44980 bytes for local variables. Maximum is 81920 bytes.
Erasing 0x100000 bytes starting at 0x00000000
Uploading 375520 bytes from /var/folders/sm/tgqgjcwx3fd3zj4_fh5nq4tw0000gn/T/arduino_build_782497/esp3d.ino.bin to flash at 0x00000000
................................................................................ [ 21% ]
................................................................................ [ 43% ]
................................................................................ [ 65% ]
................................................................................ [ 87% ]
...............................................                                  [ 100% ]
[SPIFFS] data   : /Users/ken 1/Desktop/ESP3D-master/esp3d/data
[SPIFFS] size   : 512
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/404.htm
/favicon.ico
/index.html
/index.html.gz
skipping .DS_Store
SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!

Seems like the master version can not transfer the data files....

kina500, 2018-03-16:
Update FYI

Just flashed V0.9.99 with the new index.html.gz from your link. The index file I placed in the data folder before using the 8266 sketch uploader tool in Arduino IDE.

Now the macro works.....
But I guess I still cannot update the index file with the webui.

kina500, 2018-03-16:
Very strange....
After refreshing the webui page, the macrocfg.json is 0 byte (empty).
So macro don't work in 0.9.99

luc-github, 2018-03-16:
sorry what size of SPIFFS are you using ? and what size of flash are you using ?

kina500, 2018-03-17:
512Kb SPIFFS and 1M flash.
No flash is not full, 82KB used og 471KB, if shows.

kina500, 2018-03-17:
[image]

luc-github, 2018-03-17:
Ok I just flashed a 1MB esp01 with 512K SPIFFS and cannot reproduce your issue
Did you changed any setting in configuration before flashing ?
can you also share the version UI/FW and ESP core you are using ?
[image]

[image]

also be noted that 512K SPIFFS won't allow you to use WEB update as no space to copy the new flash unlike what is wrote on report - looks bug in space calculation for 1M flash

kina500, 2018-03-17:
I cannot share the master firmware version, as I could not transfere the SPIFFS files, as I pasted in previous post, only firmware I can flash and also succeed in transferring the SPIFFS files is 0.9.99.
In 0.9.99, I can also transfere the newest Index.html.gz.
I did not change anything in settings or files before flashing and transferring files.
UI is 0.9.82LE, the one you linked to in previous post

[image]

luc-github, 2018-03-17:
Please try update with latest git because there are several fixes for Macro and also set CPU to 160Mz 
there is not change for SPIFFS between 99 to latest

luc-github, 2018-03-17:
uploading new flash should not modify existing SPIFFS if you set same size

kina500, 2018-03-18:
Updated. Info shows: UI: 0.9.82LE / FW: 0.9.101
Macro works as long as I don't close down webui and open again, then the macro is lost.
No macrocfg.json file in SPIFFS, I uploaded the macrocfg, created the macro, and the macrocfg file disappear again....

luc-github, 2018-03-18:
what you mean by upload the macrocfg ?
you just need to click macro editor - change what you need on it and click save - this will save the modified macrocfg.json

kina500, 2018-03-18:
If I flash master, I cannot upload SPIFFS files.
On 0.9.99 everything go well, I can flash firmware and upload SPIFFS files.
Tried to make combination, copied the master data files to 0.9.99 data folder, then it cannon upload the SPIFFS files again.........
Can you flash master firmware and use the ESP8266 sketch data uploader tool in Arduino IDE to upload master data files?

luc-github, 2018-03-18:
what you mean you cannot upload files ?
I never use the upload plugin for spiffs as FW has embedded spiffs file manager

kina500, 2018-03-18:
The macrocfg.json was disappaird....

kina500, 2018-03-18:
I follow the video on how to install the ESP3D: https://www.youtube.com/watch?v=pJGBRriNc9I
As instructed on your wiki page.
Then I flash the ESP8266. The SPIFFS don't contain any file after flash of firmware.
Then I use the menu point in Arduino IDE: 'ESP8266 sketch data uploader tool' to upload data files to SPIFFS, as instructed in the video.
I tried only to flash, as it is also stated on your wiki page that 0.9.99 contain the data files and instructions about uploading data files to SPIFFS don't need to be done anymore.
The video and the step instructions on wiki are different......
If I do that last thing, assuming I did not need to transfere (upload files), then I cannot log on to ESP8266 anymore, I guess the index file is not there.

Look like there is some misunderstanding how to flash this and get files into the thing, could you make a step by step instruction that are correct on newest version?

luc-github, 2018-03-18:
Did you read the readme ?https://github.com/luc-github/ESP3D#installation
the embedded uploader is described

luc-github, 2018-03-18:
if no index.html.gz is present the self uploader is displayed on web browser - if not it means you may need to format SPIFFS

luc-github, 2018-03-18:
I suggest you also to change SPIFFS size to 144Kb to be able to use Web update

luc-github, 2018-03-18:
I have tested using pluggin without issue - I think your SPIFFS is corrupted and need to be formatted

kina500, 2018-03-20:
Sorry for my late reply.
Thank you very much for your support. You made an incredible work with your software, I also used your Repetier for XYZ Da Vinci earlier. 
Something just happened in my family, so I will not be able to continue my 3D printer project for long time.
I think you can close this issue, maybe it is just my setup or use....
If I get time to work on my project again at some time, I guess ESP3D and WEBUI has changed so much I have to start from beginning again.

luc-github, 2018-03-20:
thank you - sorry to read this, I wish you everything is fine soon

Take care

closed 2018-03-20 by luc-github