luc-github/ESP3D-WEBUI #23: How to build?

luxarts, 2018-03-23:
Hi!
Great work with the firmware and Web-UI.
I want to modify the web interface a bit to complement with my printer's theme. How can I build my web into .gz file? I saw the index.html.gz file and it's only an html with the scripts and css. I don't know if I have to copy the js and css content into the html file or I can use the complete structure compressed.

I'm very lost with this, I would appreciate any help.

luc-github, 2018-03-23:
the recipe is in the wiki : https://github.com/luc-github/ESP3D-WEBUI/wiki/Compilation---Development

luxarts, 2018-03-23:
I can't believe two thing:
1) I didn't look in the wiki, I feel very stupid now.
2) The fast response, thanks!

luc-github, 2018-03-23:
no problem - have fun 😄

closed 2018-03-23 by luxarts

luc-github, 2018-03-23:
FYI I have just added a travis script for online compilation test - it may not be optimized but it works ^_^