luc-github/ESP3D-WEBUI #26: Can't run gulp

luxarts, 2018-07-20:
Hello!

I'm trying to run your gulp file. I folllowed all the steps froma clean install but when I try to run "gulp package" from the command line in the directory I get "Error: Task requires a name that is a string".
I fixed it removing the lines like `gulp.task(clean);` but now I get a new error, "TypeError: gulp.series is not a function". I found that it's because gulp.series was included in gulp 4.0.0 and when I run `npm install --save-dev gulp` I get 3.9.0.
These were the steps you follow to install and compile it? If not, can you fix it in the wiki?

Thanks!

luc-github, 2018-07-21:
I am guessing you use old node, be sure you use latest LTS one (V8.11.3)

luxarts, 2018-07-21:
I'm using node 8.10.0 and npm 5.6.0

luc-github, 2018-07-21:
Did you reinstall everything after updating node to 8.11.3 ?

luxarts, 2018-07-21:
Yes, I ran the install.bat file.

luc-github, 2018-07-21:
here what I have : 

```
E:\github\ESP3D-WEBUI>node -v
v8.11.3

E:\github\ESP3D-WEBUI>gulp -v
[22:17:03] CLI version 2.0.1
[22:17:03] Local version 4.0.0-alpha.3
```

luc-github, 2018-07-23:
weird as this what I use to install - and I installed it several times

luc-github, 2018-07-23:
Guld need to be installed local and global