luc-github/ESP3D-WEBUI #27: Hide "wait" status

Zefram88, 2018-08-07:
On command's prompt, it shows up continuously the "wait" message, should be nice to be able to hide it.
[image]

luc-github, 2018-10-02:
I need  to sync ESP3D with 2.1, it is part of it

Zefram88, 2018-10-02:
I've updated the UI to 2.1, so I need to update firmware too?

luc-github, 2018-10-02:
Yes Web UI 2.1 version use heart beat that is not yet implemented in ESP3D stand alone  - but only on embedded FW GRBL_ESP32 and Marlin_ESP32,  need  to implement this feature as well as a way to disable it in Web-UI  which will be done soon
If you update to 2.1 it should show a disconnect windows after 20s

luc-github, 2018-10-02:
Seems I have anticipated the 2.0 compatibility - ping start only after first ping - so if FW never send a ping check never start - so disconnection should not happen  with current version of WEB UI 2.1
I have already prepared a version with check box in preferences but need to do the translations - will push tomorrow if no issue raised

luc-github, 2018-10-02:
if you uncheck verbose mode you should not see `wait` neither temperature/position feedback until you check again
[image]

luc-github, 2018-10-05:
Seems it was in my repo but not pushed - it is the case now https://github.com/luc-github/ESP3D-WEBUI/commit/af2ef654a064a0af4af770c14c60c855f4b7ee13

@Zefram88 you can check - also WEB-UI 2.1 should works with ESP2.0 - just be sure connection monitoring is not checked in preferences.

[image]

Zefram88, 2018-10-05:
That's great! Thank you

luc-github, 2018-10-06:
Can issue be closed ?

closed 2018-10-06 by luc-github

Zefram88, 2018-10-06:
Sure, the wait status is no more visible.