luc-github/ESP3D-WEBUI #31: SD Card access during printing

ModMike, 2018-08-25:
Bonjour Luc,

Apres avoir consulter le schematic du Azteeg X5 Mini Wifi j'ai determiner que le MCU a accès au SD par default mais le ESP8266 contrôle la switch.

Peut tu me dire si le contrôle d'accès a la carte SD et barre quand Marlin imprime?

Dans d'autre nouvelle,

Roy ma confirmer qu'il ta parler pour le beta test de 2.0.  Est ce que ta reçu la confirmation?  Si non, je peux t'envoyer une copie de son courriel.  Peut tu me donner un lien pour que je le telecharge?

Laisse moi savoir, merci!

luc-github, 2018-08-25:
Yes by default Marlin will have full control of SD - it is up to ESP3D to switch to get access to SD or not.

So to answer it is really better not to parse SD from web UI when printing because it will cut even very quickly the SD access from Main board, 

2.0 has a read only mode that can be set by default to avoid this - I am still thinking how to add an automatic check to see if printing is ongoing or not, to allow (or not) to disable the read only mode

I have read the discussion on Marlin github, about SD - the mecanism is transparent for Marlin and 2.0 work already with RE-ARM + Specific daughter boards using the second SD card reader present on Screen like Viki2 or GLCD. So no need to focus on Azteeg wifi specificity with ESP3D - it is not the problem.

About 2.0 - I got no feedback from Roy- If he did, it was not with me, last time we talked (16th of August) it was about Japan - ^_^
I am currently working on Marlin_ESP32  and GRBL_ESP32 - did not update ESP3D 2.0 with latest fixes but they are mostly for ESP32 so version I sent to Roy  is almost the latest.

ModMike, 2018-08-25:
Can I have your email address to forward his email?

ModMike, 2018-08-25:
Not sure what you mean by parse.  When i select a file to print does it not just issue a command to Marlin to print the selected file from it's SD card?

luc-github, 2018-08-25:
support_at_tech-hunters_dot_com

I mean refresh SD card content in Web UI
Sorry I may misunderstood : Peut tu me dire si le contrôle d'accès a la carte SD et barre quand Marlin imprime?
because not clear what does it mean

Yes print command just do `M23 /path/file.gco` to select the file then `M24` to launch print , ESP3D does not access to SD as no need at this step

ModMike, 2018-08-25:
I understand now.  So it seems that the only access is:

1) When logging in the first time
2) When uploading a file
3) When deleting a file
4) When refreshing the directory
5) When selecting a job to print (actually not accessed, only command sent)

Correct?

luc-github, 2018-08-25:
1 - on latest Web_UI 1.0 no access to SD at login
2 - yes
3- yes and also when creating directory
4 -yes
5 - no access to SD at all -  just send GCODE to printer

luc-github, 2018-09-04:
I guess issue can be closed

closed 2018-09-04 by luc-github