luc-github/ESP3D-WEBUI #34: Auto Macro

ModMike, 2018-10-18:
Started playing with macros and they are very cool.  

I need to send a G-Code every time the host starts, is there any way to do that?

Also is there a list of all the ESP commands I can put in a macro?

luc-github, 2018-10-18:
the macro are part of webUI so not sure it is best way
what do you call the "host start" the printer FW ? the remote host? 
the public commands are : https://github.com/luc-github/ESP3D/blob/master/docs/Commands.txt

Azteeg WiFi has more commands releated to its features

ModMike, 2018-10-18:
Isn't it 4:00 am for you?  You sure are dedicated!

One of the limitations of the Marlin implementation is that the SD card is not auto mounted, making Auto power off resume impossible.  I want to send an M21 code every time the board has finished booting.  Can I stick it in somewhere?

luc-github, 2018-10-18:
Nope I am in Taiwan now so 10:50 am ^_^
Yes I see the discussion. 
Does not Marlin has a auto sd filename ?
Also are you sure the SD detect pin does not go to also to main mcu. As it should be a shared pin between azteeg and esp

ModMike, 2018-10-18:
I checked the schematics on dip trace and it goes straight to ESP8266, so no SD detect, that I know of, unfortunately.

I will recheck dip trace.

luc-github, 2018-10-18:
the SD detect tel smoothieware to read SD content again if modified but I do not have azteeg wifi so I cannot confirm sorry

luc-github, 2018-10-18:
this should be a part of Marlin FW - that said adding an auto macro file when esp3D start is possible but also dangerous if happen when no expected like during a print

ModMike, 2018-10-18:
I am not sure I understand, how can the machine be printing if the controller is rebooting?

luc-github, 2018-10-18:
but you can add custom init in Marlin 2.X it should solve your problem
in your HAL :
```
#define HAL_INIT 1
void HAL_init(void);
```
and in your HAL_init() in cpp do what you need

ModMike, 2018-10-18:
I could but it would never be merged, meaning future users will be in the cold.  If we do it via WEB UI, every X5 Mini customer has it.

luc-github, 2018-10-18:
if you add a #define for the board it limit to the board targeted

IMHO you could push a solution like :
#FORCE_SDMOUNT 1

in configuration.h

and force SD mount at boot if define is set

Not only Azteeg board should have this problem

I am surprised this not already available

luc-github, 2018-10-18:
anyway I have added it in todo list https://github.com/luc-github/ESP3D/issues/270

that said there is already an option at start for azteeg wifi to check update on SD - so just need to M21 before doing it - it is quick and dirty but should be ok

luc-github, 2018-10-19:
As I moved issue to ESP3D I guess this topic can be closed

closed 2018-10-19 by luc-github