luc-github/ESP3D-WEBUI #39: SD Card filesystem to a ESP8266 (Upload / Download) from http server

locomx, 2018-12-04:
@Im realy new in 3d printing, so meaby sound crazy this.
But i use a arduino sketch form other person for download and upload files from a SD Card connected to a ESP8266. 
"Using HTTP and an HTML interface to Upload and Download files to an ESP32 or ESP8266 Filing System"  - https://github.com/G6EJD/ESP32-8266-FIle-Upload
Here is the demo for this code - https://www.youtube.com/watch?v=1OkEXeRbq_M&t=170s

So the idea is do not let mi laptop wating to the printer finish the job. Just upload the files in the SD Card and print in any time that you need or let other files for history or re-print in other moment.

luc-github, 2018-12-04:
Your SD card reader is connected to printer ? Or only to esp board ?

locomx, 2018-12-04:
At moment that i start to think in this its in the ESP.

luc-github, 2018-12-04:
So how printer can access to the uploaded file?

locomx, 2018-12-05:
`If the ESP3D is connected to the serial port of the printer or to the Mega 2560

luc-github, 2018-12-05:
So esp is acting as host ? Or esp need to transfert file from esp8266 sd to printer sd ?

luc-github, 2018-12-08:
@locomx any update ?