luc-github/ESP3D-WEBUI #45: Polish translation

ArturNadolski, 2019-03-24:
I'm not proud of it but it's good start point. Maybe someone else will polish Polish transalation ;)
Uploading only changed files:
[js.zip](https://github.com/luc-github/ESP3D-WEBUI/files/3000300/js.zip)

luc-github, 2019-03-24:
Yes translating is not always easy - just one thing how do you write `Polish` in polish ?

ArturNadolski, 2019-03-24:
Polish=Polski

ArturNadolski, 2019-03-24:
What about translation of Oled messages?

luc-github, 2019-03-24:
about oled translation - I did not prepared the API yet - as this part is in FW I think it will be a standalone package on SPIFFS  - but I did not  thinked about details yet - will be in 3.0 for sure

luc-github, 2019-03-24:
will push your translation later today

ArturNadolski, 2019-03-24:
BTW.. You can add this link to the description of  translation procedure:
https://pl.wikipedia.org/wiki/Kodowanie_polskich_znak%C3%B3w
For Polish languege please use last row from the table.

luc-github, 2019-03-24:
https://github.com/luc-github/ESP3D-WEBUI/commit/b1407c01bd3b536e2fc3d8ad15c71f0b406651dc 
Merged - thanks you

ArturNadolski, 2019-03-28:
[pl.zip](https://github.com/luc-github/ESP3D-WEBUI/files/3019215/pl.zip)
update

luc-github, 2019-03-28:
Will check later

luc-github, 2019-03-28:
Merged https://github.com/luc-github/ESP3D-WEBUI/commit/e107364ca76cc54726d47186ccfff87b07604187

thank you

closed 2019-03-28 by luc-github