luc-github/ESP3D-WEBUI #47: Decrease WebUI size Ideas for 2.X and 3.0

luc-github, 2019-03-26:
Thanks to you now Web UI is getting a lot of language support and I guess more are coming 
With this benefit come a drawback - size is growing. 
On ESP32 the SPIFFS size is 165.70 KB if bluetooth is enable to be able to keep web update, because bluetooth use a lot of flash. So soon the SPIFFS will be overloaded and I suspect it is already the case with latest languages added.
So need to reduce webui size:
I think use ID instead of UK sentence will decrease size of language pack by 25% as no more UK but just an ID, but this will be ok only for a while - need to find another API  for language pack

* A - use a configurator to build the webui according language and options
PRO: could decrease the webUI size a lot depending language / options
PRO: could be applied for 2.X and 3.x as mo API change
CONS: need to build UI if want to change the language list
CONS: not sure this configurator can be done on github

* B - use english by default and load a language pack present on SPIFFS at start.
could be UK + language pack or all language packs present in some specific directory, not sure if limit to 1 language pack is a way to go, each language pack will be tar.gz
PROS: could decrease the webUI size a lot as only loaded limited number of language
CONS: Need to add manually each language to SPIFFS
CONS: Only for 3.X as big change at start and file creation

Any feedback is welcome, for the moment I I am more for the B as I am sure not to break the existing one

just-jason, 2019-03-26:
You have made a great web update feature that makes firmware updates so easy. Making config changes and compling/building a new firmware is super fast, and the update is easy with the webUI. I could see the SPIFFS being used for  better things than storing language packs. My vote would be for A.

luc-github, 2019-03-27:
@just-jason thank you - the major issue for A is : It needs to host a configurator application and I think github does not allow it out of the box, may be have some tricks but did not find yet

just-jason, 2019-03-27:
Not sure if I am uderstanding your plan clearly. 
for option A : Would you be using selectable languange options in the source code configuation ( eg https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/language.h#L35 ) ?

luc-github, 2019-03-28:
Humm not really - My idea was to provide an index.html.gz already built like today, today you only need to rebuild if you want to customize something.
what you describe could be an option C which is also a possibility but not my prefered as user need to setup build environment on his side, and because several people  already have issue just setup arduino and compile ESP3D, I can imagine how difficult will be for them to setup nodejs, gulp etc...
My idea for A was more your option C but build on Github and let user download the custom index.html.gz. Because compilation is fast unlike ESP3D FW itself it should be ok.

aganov, 2019-03-28:
travis-ci build with github releases should do the trick https://docs.travis-ci.com/user/deployment/releases/

```
index.html.gz
index.en.html.gz
index.bg.html.gz
```

luc-github, 2019-03-28:
Yes this would work, but for dual language only (english + another) , not if you want more than 2
I already use this solution here https://github.com/luc-github/Repetier-Firmware-4-Davinci/releases
but it may be also a solution

luc-github, 2019-10-04:
Becuse multilanguage file is now too big for 128Kb flash , I have added a switch to allow to build with only one additional language:
`gulp package --lang en` will only build english
`gulp package --lang fr` will only build french + english
`gulp package --lang es` will only build spanish + english
`gulp package --lang it` will only build italian + english
`gulp package --lang de` will only build german + english
etc...

I have created a batch that build all files and put the result in language directory

This is for 2.1, I hope to find a better solution for 3.0

luc-github, 2020-03-28:
I was thinking to add in the 3.0 a feature in wizard that allow to download a `profile` from github with settings to be applied to UI like  : Printer ABC, Printer XYZ, printer ABC settings will be Marlin / 2 extruder / bed /fan control, etc... this will update automaticaly ESP3D settings and ESP3D-WEBUI settings, so not need to go to settings to setup everything, even it can be done later

So back to language pack splited out of webUI, in same spirit I am thinking to put the language pack in same kind of above process, and can even be tailored : e.g no 3D printer translation in GRBL and vice-versa, all language pack files hosted in github.

This will save space, because today users need one language for one target FW, not all, but they want to choose easily the first time, so no need to keep every language/target in WEBUI all the time, english will be the only one embedded.

Using github as server will allow to keep list updated without updating index.html.gz just download the new/updated langage pack. this can be do manually of course for user not allowing internet access but again, this need to be done only once in theory

Any pro / cons about this approach ?

luc-github, 2020-05-01:
per @d-a-v use gzip binary FW is a possible improvement, it need latest ESP8266
it need to be tested -TBC

luc-github, 2020-05-23:
use compressed binary from 2.7.1 seems working 🎆  OTA is working on 1MB flash and one language pack

Language pack are unitary in 2.1 and need less than 100KB 
On 3.0 the languages pack are cumulative so it will fit also when using english only or one language pack

3.0 is not done yet but splitting GRBL/3DPrinter code and translation make code and language pack smaller  so it should be ok 
I think this thread can be closed now

closed 2020-05-23 by luc-github