luc-github/ESP3D-WEBUI #49: Can't see temp bed

ellmotion, 2019-03-29:
Hello,

I have ESP3D 2.1 and ESP3D-WEBUI 2.0 and I didn't see my bed temp.
I have try on ESP3D 2.0 and same issue.

My printer run on Marlin 2.0.
[image]

`M105
ok T:20.00 /0.00 B:20.00 /0.00 @:0 B@:0`
How can I solve this ?

Thanks in advance
(bigtreetech skr 1.3 run in 32bits)

ellmotion, 2019-03-29:
Hum.. Sorry forget this...
I have found it .. I got misled by translation

closed 2019-03-29 by ellmotion

luc-github, 2019-03-29:
Oui desole `lit chauffant` ca me fait bizarre `plateforme chauffante` c est plus coherent

DROID8622, 2019-08-08:
I cannot find it too, where can I enable hotbed temp?

fdistorted, 2020-03-08:
Maybe I am a bit stupid but it took me 20 minutes to find how to set it up.
Here is `Preferences` button location
[image]
and here is bed temp checkbox
[image]
hope it will help somebody

luc-github, 2020-03-08:
I may add these to setup wizard in 3.0

luc-github, 2020-03-08:
@fdistorted you use old version 
2.1b56 has this :
[image]

kojderek, 2020-05-02:
Hello 
I got the similar issue. 
Webgui shows e0 temp but does not shows bed temp (no values and blank graph )
[image]

luc-github, 2020-05-03:
this is different issue - please open new bug issue and fill template