luc-github/ESP3D-WEBUI #52: Apple captive portal break UI

luc-github, 2019-04-14:
* Javascript is limited 
* File browser is disabled in CNA but drag and drop is OK
* Websocket is disabled
* cannot open browser page from CNA

Solution : detect page is http://hotspot-detect.html and display a message like :
`Please open http://<ip> in your favorite browser`

luc-github, 2019-04-22:
So far seems not easy to detect captive portal on apple
* Using URL is not good solution : http://captive.apple.com works in AP mode in browser
* Using what is failing is the solution but not that simple:
 - File upload just do not answer and need user to start an upload to detect
 - WebSocket is create successfully but after raising error event which unfortunatly has code and reason `undefined`
- WebSocket disconnect with no reason but with code 1006 which may not be specific to captive portal
- WebSocket does not raise connect event

Solution => need to grab several clues:
1 - Websocket never raise connect event 
2 - Disconnect event code is 1006
3 - Error event is raised instead of connection event
4 - CurrentID message is never received 
5 - OS is OSX in browser user agent

I think if these `flags` are all detected, we can safely consider we are in captive portal of Apple and show a message explaining user that he to open a real browser.

I have only an old MacAir with Sierra OSX to validate, no Ipad neither Iphone, if anyone has a better idea - I would appreaciate 
 
Here the page I did to check because console.log is not available in captive portal window

[index.zip](https://github.com/luc-github/ESP3D-WEBUI/files/3103819/index.zip)

Nathan-ma, 2019-08-11:
Good Topic, I'm trying to use websockets on captive portal with also no success. Hopefully we can find a solution

Nathan-ma, 2019-08-11:
Actually it just worked to me here...
I used this:
`var connection = new WebSocket('ws://192.168.4.1:81/', ['arduino']);`
instead of this:
`var connection = new WebSocket('ws://' + location.hostname + ':81/', ['arduino']);`
 its working fine here

luc-github, 2019-08-12:
Interresting , so using ip address is working. Your AP ip is actually 192.168.4.1 ?

Nathan-ma, 2019-08-12:
yes, my board IP address on AP mode is actually 192.168.4.1, and pointing to this address actually make the connection work both on captive portal and web interface on mobile, I don't know about STA yet, but i'm guessing it might, well if not, I think I can use both Websocket connections on the jscript code.
Worth the shot right?

luc-github, 2019-08-13:
yes it make sense in AP mode name resolution for hostname is not always working, so use IP for AP mode and location.hostname for STA mode should be ok and cover the IP lease change in STA mode

Nathan-ma, 2019-08-14:
well glad I could help

luc-github, 2019-08-27:
@Nathan-ma I have implemented the IP set in 3.0 per your suggestion

luc-github, 2020-09-12:
Seems now on android the upload button is disabled so need also a way to detect the captive portal on android to display message

luc-github, 2020-09-15:
ok after more research and tests I still cannot find a 100% reliable way to detect captive portal, seems feature evolve and no common way, testing localstorage / sessionstorage is not working, 
so I think I will rely on this article data (https://success.tanaza.com/s/article/How-Automatic-Detection-of-Captive-Portal-works)
which give me this function
```
function isLimitedEnvironment() {
    const sitesList = [
        "clients3.google.com", //Android Captive Portal Detection
        "connectivitycheck.",
        //Apple iPhone, iPad with iOS 6 Captive Portal Detection
        "apple.com",
        ".akamaitechnologies.com",
        //Apple iPhone, iPad with iOS 7, 8, 9 and recent versions of OS X
        "www.appleiphonecell.com",
        "www.itools.info",
        "www.ibook.info",
        "www.airport.us",
        "www.thinkdifferent.us",
        ".akamaiedge.net",
        //Windows
        ".msftncsi.com",
        "microsoft.com",
    ]
    for (let i = 0; i < sitesList.length; i++) {
        if (document.location.host.indexOf(sitesList[i]) != -1) return true
    }
    return false
}
```

I will use this test to display a message to ask user open page in browser due to limited environment - I will add this in embedded page and in webUI

If anyone has better idea - feel free to suggest

luc-github, 2020-09-22:
I have added the function in embeded page and if captive portal is detected it show the message:
[image]

I will add same kind of information in index.html.gz

luc-github, 2020-09-27:
Message is now implemented also in webUI

closed 2020-09-27 by luc-github