luc-github/ESP3D-WEBUI #53: No Files in SD Files tab

mmoustache, 2019-04-18:
Hi! I have 2.0 and 2.1 versions of ESP3D. There is no Files in SD Files tab. I checked ls command and files list are OK. Running on Smoothieware.

[image]

[image]

luc-github, 2019-04-18:
Looks like the refresh button is missing I will check that
What is [ESP800] output?

mmoustache, 2019-04-18:
where do i check ESP800 output?

luc-github, 2019-04-18:
In command panel send [ESP800]
ouput is in same place like you just did ls

mmoustache, 2019-04-18:
[image]

mmoustache, 2019-04-18:
[ESP800]
FW version:2.0.0.c25 # FW target:smoothieware # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MYESP

mmoustache, 2019-04-18:
I debugged code, but there is no result. Something more the visibility of button

[image]
[image]

luc-github, 2019-04-18:
Yes will fix this tomorrow it is 11pm here
 some test should be buggy if refresh button does not display the action also may have pb with action too

luc-github, 2019-04-19:
ok looks like I was missing something reset does not set the SD target which is ok but for smoothieware 

>[ESP800]
FW version:2.0.0.c25 # FW target:smoothieware # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MYESP

but on my side EEPROM is already updated so I did not saw the problem during my tests and missed this case, my bad

short therm solution :
enter this should solve the issue by adding manually the missing info for current webUI:
```
[ESP401]P=851 T=B V=1
[ESP401]P=852 T=B V=2
```

Permanent solution:
I will update WebUI to handle `FW target:smoothieware # FW HW:Serial SD # primary sd:none # secondary sd:none` properly

luc-github, 2019-04-19:
index.html updated : https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz
please confirm any of previous solution works for you - tested Ok on my side

mmoustache, 2019-04-19:
Thanks, works for me :)

[image]

luc-github, 2019-04-19:
Great I close issue then

closed 2019-04-19 by luc-github