luc-github/ESP3D-WEBUI #55: Review Wizard

luc-github, 2019-04-27:
May it is time to review Wizard steps and behavior for 3.0
it was suggested to propose Extruder / bed configuration by @mat2k

Another solution would be to simply open preferences at the end of wizard as currently wizard only change ESP3D settings not WebUI ones but language

Any suggestion is welcome

mat2k, 2019-04-27:
What does the "Printer configuration" tab do? I see it empty (is it just me?), but I guess it'd also be a good place to add printer-related settings like number of extruders or power supply shutdown capability, etc

Putting all of that on the wizard would also make sense, since it'd be better if you could set ESP3D up and start using it right away

also, a way to get a checksum of the sdcard file would be nice - my second print didn't finish sending (abt 105 kB out of the 150 kB), but the printer reported successful transfer anyways. This would be more related to ESP3D but there would be necessary ui elements too

luc-github, 2019-04-27:
Printer Configuration allow to edit printer EEPROM settings in REPETIER and Marlin, config.txt on smoothieware
for Marlin 1.1.9 it edit what command M503 give :
```
M503
echo:  G21    ; (mm)
echo:  M149 C ; Units in Celsius

echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
ok
```
which give in UI
[image]
So yes you have some issue as you may have this feature not enable in your Marlin Configuration.h, check what M503 give you

Checksum of file will be difficult because to save tranfert time I remove all comments,  that may explain the size difference -  also if it says transfert Ok, it means printer is happy and did not throwed error with what was transfered.

>power supply shutdown capability,

where this setting come from ? what ESP3D can do with it ?

mat2k, 2019-04-27:
> Checksum of file will be difficult because to save tranfert time I remove all comments, that may explain the size difference - also if it says transfert Ok, it means printer is happy and did not throwed error with what was transfered.

I actually realized that the file was incomplete because the printer stopped at abt 2/3rds of the piece (luckily it was not too big). After this I checked and saw the file size difference. I reuploaded the file just to test and this time it's 143 kB in size, which seems OK.

yeah, i think i don't have m503 active for some reason

And PS shutdown is just an example for another option to add, I don't use it but thought it might have already been implemented into ESP3D, it's just the ability to turn the power supply on/off (usually ATX from a PC, but some other supplies have it) via the PS_ON cable by pulling it low by a controller pin. Marlin has a gcode command for that. i'm not asking for it to be implemented but an idea would be to add a button in the webui that sends that gcode command if switchable power supply is specified in the settings

luc-github, 2019-04-27:
>an idea would be to add a button in the webui that sends that gcode command if switchable power supply is specified in the settings

for that you have macro buttons:
[image]

luc-github, 2019-04-27:
Because all printers are differents and Printer FW use different GCODE, wizard cannot cover all possible configurations
Also I know by experience, wizard must be small and as brief as possible, people do not like to read the doc, people do not like to answer too many questions - they just want to use ASAP
That is why current wizard is basic - (may be too much ?) but I do not plan to cover all settings because everything is configurable later with preferences - macro

luc-github, 2020-03-29:
Some steps in wizard:
Condition to be launched : presence or not of preferences.json and/or unknow firmware defined (Marlin_ESP32 and GRBL_ESP32 are already defined so cannot be an unique condition)
Also wizard will be different depending of wifi setup

#### if ESP is in station mode:
 * Langage selection :
UI will download the available language list from github and download only the selected language
as described https://github.com/luc-github/ESP3D-WEBUI/issues/47 
 * download and apply a profile from github with settings to be applied to UI like : Printer ABC, Printer XYZ, printer ABC settings will be Marlin / 2 extruder / bed /fan control, etc... this will update automaticaly ESP3D settings and ESP3D-WEBUI settings, so no need to go to settings to setup everything,  but of course it can be done later any time
* other TBD 

####if in AP mode
 * langage selection according what is on filesystem
 * setup wifi client mode 
 * if FW target is not set propose to define FW target (mandatory)
 * other TBD

luc-github, 2020-11-02:
I have added a flag in ESP3D EEPROM for setup - this will  allow to not rely on any  file presence neither specific settings - etc... flag is readable in [ESP800] and will be updated by ESP401 if necessary

luc-github, 2020-11-05:
Basic API for wizard is now ready - time to populate the steps
There is a button in settings in case user want to relaunch it 
[image]

[image]

the setup flag is also modified  or not when closing:

[image]

Unlike 2.1 you can navigate in steps 

the top headers may be differents for mobile view due to display size and will be adjusted soon

luc-github, 2021-10-18:
This is canceled for 3.0 another will be done for rewrited version which allow to grab control more easily and display wizard pages in full screen - so cancel this topic

closed 2021-10-18 by luc-github