luc-github/ESP3D-WEBUI #57: Show bed temperature

Hasenpups, 2019-05-12:
Hello, 

How is it possible to show the bed temperature? I couldn't find a setting for this. 

Regards
Stefan

luc-github, 2019-05-12:
Go to preferences and enable bed check box

Hasenpups, 2019-05-12:
Thank you. It is located under: "Show extruder panel"
Which doesn't make sense to me. I was expecting it to be under the section "Show temperatures panel"

luc-github, 2019-05-12:
Indeed it is confusing and not correct - I was sleeping when I did that I think and until now no one repported it 
[image]

[image]

I will move that to right place - thanks for reporting

luc-github, 2019-05-12:
fixed in https://github.com/luc-github/ESP3D-WEBUI/commit/8267c9293a4af57086419d07af44bb20e8e730c2
[image]

luc-github, 2019-05-12:
thank you @Hasenpups  for the report

closed 2019-05-12 by luc-github