luc-github/ESP3D-WEBUI #58: Translation: some text is hardcoded and can't be translated

SergioUno, 2019-05-19:
This text is not included in translation file, but it shown it the tab "GRBL config":

    "$0":"Step pulse, microseconds",
    "$1":"Step idle delay, milliseconds",
    "$2":"Step port invert, mask",
    "$3":"Direction port invert, mask",
    "$4":"Step enable invert, boolean",
    "$5":"Limit pins invert, boolean",
    "$6":"Probe pin invert, boolean",
    "$10":"Status report, mask",
    "$11":"Junction deviation, mm",
    "$12":"Arc tolerance, mm",
    "$13":"Report inches, boolean",
    "$20":"Soft limits, boolean",
    "$21":"Hard limits, boolean",
    "$22":"Homing cycle, boolean",
    "$23":"Homing dir invert, mask",
    "$24":"Homing feed, mm/min",
    "$25":"Homing seek, mm/min",
    "$26":"Homing debounce, milliseconds",
    "$27":"Homing pull-off, mm",
    "$30":"Max spindle speed, RPM",
    "$31":"Min spindle speed, RPM",
    "$32":"Laser mode, boolean",
    "$100":"X steps/mm",
    "$101":"Y steps/mm",
    "$102":"Z steps/mm",
    "$110":"X Max rate, mm/min",
    "$111":"Y Max rate, mm/min",
    "$112":"Z Max rate, mm/min",
    "$120":"X Acceleration, mm/sec^2",
    "$121":"Y Acceleration, mm/sec^2",
    "$122":"Z Acceleration, mm/sec^2",
    "$130":"X Max travel, mm",
    "$131":"Y Max travel, mm",
    "$132":"Z Max travel, mm"

luc-github, 2019-05-19:
well the text is not hard coded it can be translated if you look at the code : https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/www/js/config.js#L504-L546
` inline_help(label)` convert the code to english then translate the english to active language selected
```
function inline_help(label){
    var shelp = "";
    shelp = grbl_help[label];
    if (typeof shelp === 'undefined' ) shelp = "";
 return translate_text_item(shelp);
}
``` 
if is not currently translated it is because it is not actually translated - but you actually point out another issue, the english text is not in  `lang.tpl`  so it will be missed when translators will work on it. and it is indeed
I will add it

luc-github, 2019-05-20:
I have added the missing lines to language.tpl https://github.com/luc-github/ESP3D-WEBUI/commit/299c2cd3bdb118da806e15a17d7f86c3940cda03
and I have updated the translation status : https://github.com/luc-github/ESP3D-WEBUI/wiki/Translation-support
thank you for finding this

closed 2019-05-20 by luc-github