luc-github/ESP3D-WEBUI #64: Add fan speed control

just-jason, 2019-06-22:
**Is your feature request related to a problem? Please describe.**
No, it is not a real problem, just a nice extra

**Describe the solution you'd like**
Add the possibility to control fan speed just like the option to control flow or print speed

**Additional context**
I have found myself using this a few times on prints with large area. Most slicers will turn off the fan for the first layer. If the first layer is very large, then the extruder can suffer heat creep and  block up because it is not getting any cooling for a very long time. I have always done this by sending G-codes commands like M106 S20 through ESP3D, and I think it would be a nice addition to have a similar system like to flow control or print speed control for this.

luc-github, 2019-06-22:
sorry did you check preferences ?
[image]

just-jason, 2019-06-22:
Mmm. No I did not. Can I ask you, was it always there .? I am not to sure if I am on the latest webui version. I will make an update anyway and check. Sorry for the false alarm.

closed 2019-06-22 by just-jason

luc-github, 2019-06-22:
Before it was displayed by default, then I put it  in preferences
So yes was always there ^_^