luc-github/ESP3D-WEBUI #65: [BUG] ESP3D restarts when print is paused

just-jason, 2019-06-22:
**Describe the bug**
When pausing a print, ESP3D reboots, and a new login is necessary

**To Reproduce**
Steps to reproduce the behavior:
1. Start print
2. Pause print ( eg to change filament)
3. Check WebUI of ESP3D
4. See error --- Need to log back with User / password

**Expected behavior**
I would expect to remain logged in on pausing a print, 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:  (please complete the following information)**
- Version: 2.1
- Wifi mode: STA
- Any change done in FW - None in ESP3D, In marlin I am using #define ADVANCED_PAUSE_FEATURE

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version 2.0

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name: Adafruit Feather
- Flash size: 4M: 3M/1M

**Browser used (please complete the following information):**
- Name: EDEDGE

- Opereation system : Windows 10

luc-github, 2019-06-22:
what cause reboot ?
the log out is automatic if no activity during few minutes or after a reboot as no more credential in memory

just-jason, 2019-06-22:
There was no long activity pause, and , at least on the printer side I can not see any power issues. The esp is on the same 5v line as the rest of the mega so I would expect power loss to affect the whole printer, but this is just happening to me when making a pause to change filament, ie 1min max. I am also surprised to see it and I might not even have noticed it if I did not have the password enabled on the eap3d which means on I need to log back in . I will run some more teats next week and get back to you.

just-jason, 2019-06-25:
I updated the ESP3D today, and I have no longer got this issue. I am not sure what the exact cause was, but as it is now working I am closing this

closed 2019-06-25 by just-jason