luc-github/ESP3D-WEBUI #70: [FEATURE REQUEST] Please find attached the completed french translation.

ewidance, 2019-09-16:
[fr.zip](https://github.com/luc-github/ESP3D-WEBUI/files/3618380/fr.zip)

luc-github, 2019-09-16:
Merci ^_^. Je n 'ai meme pas le temps de faire le francais.

luc-github, 2019-10-08:
Missed this one

luc-github, 2019-10-08:
@ewidance sorry I cannot open your zip file - I got virus alert
[image]

Can you upload it as text only ?
Thank you

ewidance, 2019-10-08:
[fr.js.txt](https://github.com/luc-github/ESP3D-WEBUI/files/3700915/fr.js.txt)
I've dowloaded it and unzipped on my professionnal PC (fresh install). No threat inside, just a js file... and no alert. Maybe you've got polluated files in your temporary download. 
Find included text.

luc-github, 2019-10-08:
thank you 
well I do not know where the problem come from, I even cannot download the zip file
but no issue with text version
[image]

luc-github, 2019-10-08:
update done https://github.com/luc-github/ESP3D-WEBUI/commit/e700f258e79d2f1def00f0703838f56f2a6d9660

Thank you

closed 2019-10-08 by luc-github