luc-github/ESP3D-WEBUI #72: LED Support

Wuckel01, 2019-10-07:
is it possible with one
Serial Wifi Adapter Modules ESP8266 ESP-12E
adapter a relay or with a 12V 24V 36V High Voltage Switch Module 3V 5V Low Control E-Switch Mosfet Arduino to switch with a GPIO via the webUI so that you can turn the light on and off at the ducker.

luc-github, 2019-10-07:
yes - use the macro buttons and assign a macro using [ESP201] command

https://github.com/luc-github/ESP3D/blob/2.1/docs/Commands.txt

Wuckel01, 2019-10-07:
Hi
Thank you for your prompt reply
where can I find the makro buttons in the webUI to set up?

or is there an exact guide?

luc-github, 2019-10-07:
there are screenshot in readme :
[image]

here a doc done by bart 
https://github.com/bdring/Grbl_Esp32/wiki/ESP3D-Web-UI-for-Grbl_ESP32#macros

Wuckel01, 2019-10-07:
Hi

Thanks for the super fast support.

you can customize the display the buttes are a bit funny

[image]

luc-github, 2019-10-07:
what resolution do you use ?

Wuckel01, 2019-10-07:
Hi
on the PC I use 1680x1050
but even if ch the browser zoom in or zoom in Firefox change that does not really what. Man can not arrange the butts with each other.

Wuckel01, 2019-10-07:
Hi
have something forgotten on the tablet android and mobile phone is just like that.

luc-github, 2019-10-07:
I did update that should limit the issue

Wuckel01, 2019-10-07:
Hi

I downloaded the current firmware version
2.1.0 b39 is bundled and installed as bin
but nothing has changed

luc-github, 2019-10-07:
I did not updated the fw. this is webui

Wuckel01, 2019-10-08:
Hi

what do I have to copy that where?

luc-github, 2019-10-08:
copy the latest index.html.gz from this repository

Wuckel01, 2019-10-08:
Hi
great thank you so it is better everything goes and is connected
thanks for great support

luc-github, 2019-10-09:
good 😸

luc-github, 2019-10-11:
I guess issue can be closed then

closed 2019-10-11 by luc-github