luc-github/ESP3D-WEBUI #73: how do i use these macros?

thetechy2, 2019-11-01:
i am using a wifi backpack and just got it set up and now want to setup macros for preheating for different materials but have no idea where to start on making macros to quickly and easily preheat the hotend and printbed if i set the command i want to the filename it just says wrong command, if i put the gcode in the file and then save that it starts the file and says done printing file

thetechy2, 2019-11-02:
oh also any reason why my heatbed doesn't show up by default in the temperatures on the dashboard

luc-github, 2019-11-02:
For macro, here a simple doc https://github.com/bdring/Grbl_Esp32/wiki/ESP3D-Web-UI-for-Grbl_ESP32#macros

Many printers do not have heat bed, automatic detection is not implemented, so you must go to  preferences to enable it

thetechy2, 2019-11-02:
> so you must go to preferences to enable it
where can i find these settings?

luc-github, 2019-11-02:
[image]

closed 2019-11-26 by stale[bot]