luc-github/ESP3D-WEBUI #77: [Question] Connecting ESP3D ... please wait after installation

markusb, 2019-12-03:
I've installed the basic ESP3D firmware on a Wemos D1. I can connect to the AP and upload the index.html.gz for to it. After refreshing the windows I get the 'Connecting ESP3D ...' message with a moving bar and the mention 'please wait' below.

This may be because I did something wrong installing it. But I do not find any documentation about configuring/installing ESP3D-WEBUI specifically. The repository contains a script 'build.bat', but this seems to be for Windows (I am on Linux). Installing npm and gulp and running them as like the bat file suggests gets me to an error message about an operational dependency on darwin/macos. Not knowing the underlying technologies employed on the 'wrong' platform and with no documentation I'm lost here.

luc-github, 2019-12-03:
if you do not change anything, you do not need to build anything just use the index.html.gz  in this repo : https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz
 and copy it to ESP3D SFIFFS (https://github.com/luc-github/ESP3D-WEBUI#installation)

markusb, 2019-12-03:
Actually the index.html.gz file you mention is in the root of the tree of the ESP3D-WEBUI. This is also the file I did upload/copy to the ESP3D spiffs like mentioned where you said. So, it looks like I did it correctly, then.

After flashing the ESP3D firmware to the Wemos D1 it boots and I can connect to it via Wifi. I get the barebones WebUI where you can essentially upload a file to the spiffs or upgrade the firmware. This seems to indicate the the firmware works a intended.

From there I upload the index.html.gz from the root of the tree of the ESP3D-WEBUI. After rebooting or simply refreshing the page I get this 'Connecting ESP3D...' screen.

How can I debug this situation ?

luc-github, 2019-12-03:
you wrote you build it - did you ? 
> This may be because I did something wrong installing it. But I do not find any documentation about configuring/installing ESP3D-WEBUI specifically. The repository contains a script 'build.bat', but [...]

Sorry this is not clear to me

markusb, 2019-12-03:
I attempted to build it, but failed.

As fall back I used the index.html.gz provided and this index.html.gz shows the symptom.

luc-github, 2019-12-03:
Ok then please open issue with all info requested  in template
If you also can open the dev mode window of your browser and report what is the error it may also help

markusb, 2019-12-03:
In compiling the info in the form I suspect I found the culprit: I think I'm running version ESP3D-1.0 (was provided as tarball, preconfigured for my situation with a DaVonci-1.0a). I'm in the process of getting the newest version up and running. Fighting with libraries right now.

markusb, 2019-12-03:
I can confirm that this is the problem. After upgrading to the latest firmware on the ESP the GUI now works. Obviously it was looking to communicate with the ESP and the ESP was not up to speed.

luc-github, 2019-12-04:
great - happy you sorted it 😸 

I close topic then

closed 2019-12-04 by luc-github