luc-github/ESP3D-WEBUI #78: [BUG] G29 T does not output proper mesh

Hukuma1, 2019-12-25:
**Describe the bug**
G29 T (with Unified Bed Leveling in Marlin 2) does not display the full mesh.

**To Reproduce**
Steps to reproduce the behavior:
1. G29 T

**Expected behavior**
My bed has 15 x 15 grid. It should show the whole mesh. Screenshot from Pronterface and ESP3D below.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

**ESP3D Firmware:  (please complete the following information)**
- Version: 2.1.0.c26
- Web UI Version: 2.1b53
- Wifi mode:[e.g. STA]
- Any change done in FW

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version: 2.0.1

**Board used (please complete the following information):**
 - Printer Board: BTT SKR E3 Mini v1.2
 - MCU: ESP8266/8586 
 - Name:[e.g. NodeMCU2s] 
 - Flash size: 1.00 MB

**Browser used (please complete the following information):**
- Name: [e.g. Chrome]
- Version  [e.g. 2.0]
- Operating system [e.g. Windows 10]

**Additional context**
Add any other context about the problem here.
Developer tools output, error message, etc...

luc-github, 2019-12-26:
Does serial used for esp display full info when command is issued ? Like on main serial

Hukuma1, 2019-12-26:
How do I check that? Sorry, not too sure what that means. :/

luc-github, 2019-12-26:
Instead of esp board , connect uxb serial adapter and use pronterface  on this port

Hukuma1, 2019-12-26:
That's the first photo example. The Pronterface photo is using the USB port. And ESP3D is using the Wifi. Or am I misunderstanding?

luc-github, 2019-12-26:
on BTT SKR E3 Mini v1.2 you have 2 serial ports?
one for USB 
one for ESP

So I am saying need to check output of the Serial port used to connect ESP is correct - to ve3rify issue is on esp3d or on Serial port 
So connect the Serial port used to connect ESP to Serial to USB adapter and check if output is correct like same as USB, not using ESP board/neither ESP3D

Hukuma1, 2019-12-26:
Using defaults from Marlin 2.0.1 config for BTT board. What do you want me to change to try? :)

`/**`
` * Select the serial port on the board to use for communication with the host.`
` * This allows the connection of wireless adapters (for instance) to non-default port pins.`
` * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.`
` *`
` * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]`
` */`
`#define SERIAL_PORT 2`

`/**`
` * Select a secondary serial port on the board to use for communication with the host.`
` * This allows the connection of wireless adapters (for instance) to non-default port pins.`
` * Serial port -1 is the USB emulated serial port, if available.`
` *`
` * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]`
` */`
`#define SERIAL_PORT_2 -1`

`/**`
` * This setting determines the communication speed of the printer.`
` *`
` * 250000 works in most cases, but you might try a lower speed if`
` * you commonly experience drop-outs during host printing.`
` * You may try up to 1000000 to speed up SD file transfer.`
` *`
` * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]`
` */`
`#define BAUDRATE 115200`

luc-github, 2019-12-26:
Just what I wrote, no need to change Marlin settings neither firmware

Hukuma1, 2019-12-26:
Oh I see the miscommunication. The board I'm using is BTT SKR E3 Mini, but the wifi module is a direct connection to BTT TFT35 V3.0 display. It just plugs into the port there to enable wifi functionality. I cannot change where to plug it in anywhere else. Unless you mean I have to do something with the USB port on the TFT35 display? But I only use that for printing, so not sure what else I could plug in there to test.

luc-github, 2019-12-26:
Ok so let me clarify
Skr is conected to tft and esp is connected to tft
Usb port from skr show all G29 T output
Esp3d-webui does not show the complete output for  G29 T

How can you be sure issue come from esp3d ?
Could be the tft not doing proper bridge between skr and esp
Could be serial 2 on skr that does not send all data for any reason

So need to check Skr serial 2 with pronterface
And also check tft esp port with pronterface to control esp get all data and issue is not tff or marlin

Hukuma1, 2019-12-26:
I see. Yeah, I don't know. Those are all the correct variables though, you are right.

What I also do not know is how to "check Skr serial 2 with pronterface" or "check tft esp port with pronterface" maybe that is the issue for me moving forward. Is there a guide on how to connect using Pronterface via these other methods? I'm just using default settings in it to connect via USB to SKR.

luc-github, 2019-12-26:
For that you need serial usb adapter , like the one used to flash esp board

Hukuma1, 2019-12-26:
Alright, so yeah that's not happening. Guess we can either close this issue or if someone has those items and can check to proceed?

luc-github, 2019-12-26:
Well if it somethink related to your tft firmware version  or even marlin settings it may not be reproductible
How did your esp was flashed if you do not have serial usb adapter

Hukuma1, 2019-12-26:
Came with firmware flashed already from BTT's Aliexpress store. My next question would be how do I update the firmware (not web-UI, that I figured out) but I didn't see any BIN files online. :P

FW: 2.1.0.c26 (what came flashed from BTT store)

luc-github, 2019-12-26:
There is no bin because there are too many possible confgurations for esp3d and too many possible hardware/ boards 
You need to build esp3d firmware using sources

Hukuma1, 2020-01-04:
I don't have the equipment you mentioned to isolate the issue. Unless you do, I can't further debug apart from what I observed above. :/

luc-github, 2020-01-04:
can you check what is output using data port query and telnet client like putty ?

Hukuma1, 2020-01-05:
Did you probe the bed before trying to see the mesh data? G29 to probe it.

Good news is that it seems normal bilinear bed leveling seems to work fine if I want to see the mesh (command: M420 V) in ESP3D. :) It's probably just some weird issue with G29 T command?

luc-github, 2020-01-05:
well if one output work with one command and not with another one - yes there is big chance issue is on Marlin side somewhere 
Yes I tried to probe and got the `no mesh data` it may be wrong command I use or wrong settings in Marlin

closed 2020-01-29 by stale[bot]