luc-github/ESP3D-WEBUI #81: [Question] How does the printer configuration tab work?

Liger0, 2020-01-02:
Hi. In Esp3d tab -> Firmware, I did setup "marlin".
The printer configuration tab is blank. How can I use it?
Thanks.

luc-github, 2020-01-02:
this tab send gcode to printer firmware to get printer settings
so esp board need be connected to printer board
also on Marlin you need to enable EEPROM in FW

Liger0, 2020-01-02:
The ESP8266 (wemos d1 mini pro v1.0.0) is connected to the board and I can move axis etc. 
EEPROM is enabed in firmware.
There is nothing in the printer settings tab.
Also in the esp3d tab, if I click "printer", most of the times nothing happens and it keeps showing the network settings.

luc-github, 2020-01-02:
what is the M503 output?

Liger0, 2020-01-02:
Where do I see it? I can only send gcodes in the dashboard.

luc-github, 2020-01-02:
In dashboard on command panel, you can also send it by serial to your printer using arduino monitor of putty

Liger0, 2020-01-02:
I sent M503 by the dashboard but nothing happened.
I don't think it's a communication error since, if I for example home on the LCD, the g28 appears on the dashboard.

luc-github, 2020-01-02:
If no output it means EEPROM is not enabled in Marlin

Liger0, 2020-01-02:
[image]

It seems enabled to me.

luc-github, 2020-01-02:
What is M503 when send by serial?
What complete version of webui?

Liger0, 2020-01-02:
I don't have the possibility to link the wifi to the PC while the motherboard is attached to the module unfortunately. The webUI is 2.1-RC1 as much as I can see.

luc-github, 2020-01-02:
I suspect issue is https://github.com/luc-github/ESP3D-WEBUI/issues/76 which was fixed after so use latest git version https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

Liger0, 2020-01-03:
I did everything. Used core 2.5.2 instead of 2.6.3, and used that index file. No difference, the tab is still blank.
[image]

luc-github, 2020-01-03:
I just check Marlin 2.0.1 Release notes : Fix M503 output for MBL
so something may change again ... I will check

luc-github, 2020-01-03:
I cannot reproduce your issue
[image]
[image]

[image]
[image]

[image]

Be sure M503 give an output 
[image]

Liger0, 2020-01-03:
As I said unfortunately I can't hook the entire printer to the usb. 
Either way I can't upload files to the SD or see what is in (error 2) and I can't receive datas from the printer, as I can't see the temperature or anything when doing M503 etc.
But if i send a G28, the printer homes, and same if I use the buttons on the webUI.

luc-github, 2020-01-03:
so there is communication issue rx pin of your esp and tx pin of your printer
Why you cannot use printer USB ?
What is your board ?

Liger0, 2020-01-03:
I have an skr 1.3
Well, the problem is the printer is on another floor with a desk and I have it wired directly with the frame so I can't just take the control box apart. 
I may try to do it once I get esp3d web to work again though.

luc-github, 2020-01-03:
ho be careful serial pins can be damaged on btt board if connection is done upside down

Liger0, 2020-01-03:
I am unsure if the damage using the pins on the top or bottom would be shared or if they are unrelated, but I tried both of them and there is no difference. 
[image]

luc-github, 2020-01-03:
the pins a shared - top if for btt tft connector, bottom one is for AUX 1 connector  but pins are same

be sure your TX / RX configuration in Marlin is correct :
```
#define TX_BUFFER_SIZE 128
#define RX_BUFFER_SIZE 1024
```

Liger0, 2020-01-03:
So far, the configs seem to be:

```
// Transmission to Host Buffer Size
// To save 386 bytes of PROGMEM (and TX_BUFFER_SIZE+3 bytes of RAM) set to 0.
// To buffer a simple "ok" you need 4 bytes.
// For ADVANCED_OK (M105) you need 32 bytes.
// For debug-echo: 128 bytes for the optimal speed.
// Other output doesn't need to be that speedy.
// :[0, 2, 4, 8, 16, 32, 64, 128, 256]
#define TX_BUFFER_SIZE 0

// Host Receive Buffer Size
// Without XON/XOFF flow control (see SERIAL_XON_XOFF below) 32 bytes should be enough.
// To use flow control, set this buffer size to at least 1024 bytes.
// :[0, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048]
//#define RX_BUFFER_SIZE 1024

#if RX_BUFFER_SIZE >= 1024
  // Enable to have the controller send XON/XOFF control characters to
  // the host to signal the RX buffer is becoming full.
  //#define SERIAL_XON_XOFF
#endif
```

I am trying changing it.

Liger0, 2020-01-03:
I changed it with TX_BUFFER_SIZE 128 and uncommenting the RX one. 
The only difference I noticed is that, when sending a command via the webUI, I receive in the commands 
echo:LPC1768 (100Mhz) Initialized

However I still can't see anything in the SD tab and the Printer tab.

luc-github, 2020-01-03:
for SD be sure also your Marlin configuration is correct
Looks like your configuration has some problems

luc-github, 2020-01-03:
To test SD be sure M20 command can list SD content

Liger0, 2020-01-03:
I am just back to the starting point even after configuring those 2 options, I can move axis and manually send G28 but that's it. 
M20 and M503 do nothing and I don't receive anything on the command tab ever.

Liger0, 2020-01-03:
> 
> 
> for SD be sure also your Marlin configuration is correct
> Looks like your configuration has some problems

[Marlin.zip](https://github.com/luc-github/ESP3D-WEBUI/files/4020111/Marlin.zip)
I can't find any issues in the firmware. SD works on the LCD. But on the WebUI nothing works except axis. (so temperature etc)

luc-github, 2020-01-03:
I am not Marlin expert sorry, I have a topic open for btt but I am confident the issue you meet is not due to ESP3D but Marlin configuration

Liger0, 2020-01-03:
Is there anything to look at to see if the wifi module or board aren't damaged?

luc-github, 2020-01-03:
you need serial /usb adapter and connect it to  arduino monitor 
this will allow you to check communication is working or not

Liger0, 2020-01-03:
@luc-github I did set the jumper on the PSU mode instead of USB, powering it from the wall and connecting via USB to my PC, with the esp8266 unplugged.

I tried with arduino first: G28, axis movement and heating gcodes worked, but M503 didn't do anything, neither did M20. Exactly same issues as with ESP3D.

Then I wanted to give printrun (pronterface) a try and.. everything worked. M503 gave this:
```
Connecting...
Printer is now online.
echo:Active Extruder: 0
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:Active Extruder: 0
ok P15 B0
ok P15 B1
ok P15 B2
>>> M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 T1 D1.75
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00
echo: M92 T0 E411.00
echo: M92 T1 E415.00
echo:Maximum feedrates (units/s):
echo:  M203 X2500.00 Y2500.00 Z100.00
echo:  M203 T0 E25.00
  M203 T1 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X500.00 Y500.00 Z100.00
echo:  M201 T0 E5000.00
  M201 T1 E5000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P600.00 R1000.00 T800.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.07
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Hotend offsets:
echo:  M218 T1 X20.00 Y0.00 Z0.000
echo:Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H205 B60 F0
echo:  M145 S1 H250 B80 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:  M301 P22.20 I1.08 D114.00
echo:Z-Probe Offset (mm):
echo:  M851 X-31.00 Y-2.00 Z0.00
echo:Stepper driver current:
echo:  M906 X700 Y750 Z1000
echo:  M906 T0 E800
echo:  M906 T1 E800
echo:Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 T0 E
echo:  M569 S1 T1 E
echo:Filament load/unload lengths:
echo:  M603 T0 L430.00 U430.00
echo:  M603 T1 L430.00 U430.00
echo:Tool-changing:
echo: Z2.00
echo:Filament runout sensor:
echo:  M412 S1
ok P15 B3
```

Then tried M20:
`>>> M20
SENDING:M20
Begin file list
/GCODES/file.GCO 5327430
/GCODES/second.GCO 19773
End file list
ok P15 B3
`

It also shows the correct temperatures in the graph and I can upload the files from the PC.
I have no idea why arduino serial monitor and esp3d are having such an hard time!

luc-github, 2020-01-03:
Yes Usb port and serial port for esp are not same port so test must use same port to check properly

Liger0, 2020-01-03:
As much as I know, serial_-1 is the USB, and serial_0 is the TFT. 
As there was nothing on the TFT, I only tested serial_-1, so both arduino and pronterface read datas from there. 
The fact pronterface worked, means that the USB serial_-1 works, and arduino failing to work the same way leads me to think some libraries/core is unworking, thus causing the exact same issues on the flashed wemos.
I don't think there is a way to directly test the serial_0 tho.

Liger0, 2020-01-03:
Okay, I flashed a new bugfix version of marlin and it works! I flashed back the 2.0.1 stable to test and indeed it doesn't work. 
The only issue is the way the sd files are shown.
I have on the root of the SD the file: 
`FIRMWARE.CUR`
Then I have the folder
`Gcodes`
which contains 2 files:
`flow05.gco
Fuhrer.gco
`
but this is what is shown:
[image]

luc-github, 2020-01-04:
what is out put of M20 command ?

Liger0, 2020-01-04:
My lcd doesn't have an SD slot so of course selecting LCD, which is the only way I am able to use the webui, I can't see the SD, and M20 shows no file, just it regonizes the command and shows "start and end of files."
If I set ONBOARD, the webui out of setting axis coordinates and sending a temp input, won't work, but pronterface can see the M20 command and lists all of the files.

luc-github, 2020-01-04:
So to sum up : 
depending SD configuration in Marlin, serial for ESP board is working or not - but when serial is working the SD is not because you do not have lcd with SD card reader

it is definitively a Marlin bug or a Marlin configuration problem

Liger0, 2020-01-04:
You are right!
Marlin 2.0.1, which is the latest stable and of just some days ago, and I tried to redownload multiple times, never worked. 
Then I downloaded a mroe recent bugfix and only via LCD it worked.
Downloaded another new bugfix and it works totally!
So worth noting not to use stable 2.0.1

Thanks for all the help! ^^
Just a last question: will it be ever possible to print from the esp8266 or not?

luc-github, 2020-01-04:
just for macro for the moment

print from esp8266 flash mean esp3d need to be an  gcode host and this feature will be likely only on esp32 not esp8266 due to performance limitation with esp8266 - be a proper gcode host imply lot of check in ack and user cases
I am still struggling with it on esp3.0

Liger0, 2020-01-04:
Perfect, thank you. ^^

closed 2020-01-04 by Liger0