luc-github/ESP3D-WEBUI #82: [BUG] Filesystem looks empty after failing to upload a gcode file

Liger0, 2020-01-02:
**Describe the bug**
Filesystem looks empty after creating a subdirectory and then failing to upload a gcode file in it

**To Reproduce**
Steps to reproduce the behavior:

1. Go to ESP3D
2. Click on the folder icon
3. Create a directory "gcode"
4. Enter the gcode directory
5. Upload a gcode file
6. Fails to upload
7. Filesystem results empty and without the 2 other stock files.

**Expected behavior**
It should uplaod the file and that's it

**Screenshots**
[image]

**ESP3D Firmware:  (please complete the following information)**
- Version: 2.1
- Web UI Version: 2.1
- Wifi mode: No idea
- Any change done in FW: NO

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version: 2.0.1

**Board used (please complete the following information):**
 - MCU: esp8266ex
 - Name: wemos d1 mini pro v1.0.0
 - Flash size: 16MB

**Browser used (please complete the following information):**
- Name: Firefox
- Version: 71.0
- Operating system: windows 10

Liger0, 2020-01-02:
If I clean the browser or use another one, the files and fodlers (except the failed to upload ones, of course) reappear.

Liger0, 2020-01-02:
I tried another browsers, once it randomly showed the files again. even after cleaning all of them, I can't see the files anywhere.

luc-github, 2020-01-02:
What esp8266 core version are you using?

Liger0, 2020-01-02:
The latest 2.6.3

luc-github, 2020-01-02:
Please try 2.5.2

Liger0, 2020-01-02:
So I think I'll have to redo everything from 0.
What do I have to reset? 
Should I follow:

If you mess up a configuration you can pull down GPIO2 during reset/powerup to wipe the settings stored in EEPROM.

If behavior is not consistent, you may need to erase the full flash, for that use the esptool present in your ESP core instalation in tools directory with option --chip auto erase_flash
So in my case on git version of ESP32 under windows :
C:\Users\user\Documents\Arduino\hardware\espressif\esp32\tools\esptool>esptool.exe --chip auto erase_flash

?

luc-github, 2020-01-02:
Better is to flash using erase all sstting in ide to have clean flash

Liger0, 2020-01-02:
> 
> 
> Please try 2.5.2

You refer to this, right? 
[image]

I am unsure about where the "erase all settings is" in aurdino.

luc-github, 2020-01-02:
Yes
[In menu](https://www.google.com/search?q=erase+all+esp8266+arduino&client=tablet-android-asus-tpin&prmd=ivsn&sxsrf=ACYBGNRJ0uYZlKwlmbmppfhOwqfXaMV07w:1578004045672&source=lnms&tbm=isch&sa=X&ved=2ahUKEwiIqeKU--XmAhUHJhoKHTA-BIcQ_AUoAXoECA0QAQ&biw=962&bih=601#imgrc=RkK8g6UQk2rNuM)

Liger0, 2020-01-02:
I have redone everything using 2.5.2 instead of 2.6.3 and using the index file you provided me in #81 .
Now when I upload a gcode I see a popup correctly, but still.. it fails to uplaod the gcodes to the filesystem anyhow.

Liger0, 2020-01-02:
I did some tests and it seems that, while before nothing uploaded, now I can only upload files with a very short amount of characters. If I try to upload a file that has a long name, it fails.

Liger0, 2020-01-03:
I ecnountered another bug. it just loops the directory when entering one. So it keeps displaying the same files/directory and updates the directories like there are infinite equal subdirectories.

[image]

luc-github, 2020-01-03:
I cannot reproduce with latest web-UI
[image]

[image]

But it is on ESP32 with  Marlin FW not ESP8266 ESP3D FW - I will try on wemos d1 mini pro v1.0.0 as I have one

luc-github, 2020-01-03:
I cannot reproduce issue
[image]

[image]

[image]

please share ESP3D settings and FW versio you are using

Liger0, 2020-01-03:
I use Marlin 2.0.1 stable.
For ESP3D... it is strange as of today. I did set a static IP of 192.168.1.8, but today when I tried to connect nothing happened on any browser. So I checked with arduino serial monitor if the IP changed.. but it is the same. Still the web page seems not working. Is there any issue with the site?

luc-github, 2020-01-03:
I would say your FS is corrupted
you need to format it - there is an ESP command to do it

Liger0, 2020-01-03:
I think it is 
[ESP710]FORMAT pwd=<admin password>
and that I have to send it via Arduino serial monitor.
I tried using
[ESP710]FORMAT pwd=admin
as I didn't do any custom configuration but it says "Incorrect Cmd". What password does it need?

Liger0, 2020-01-03:
Okay I did format it correctly. Should I do anything else to reconfgiure it? because the ip is still the same but it doesn't work yet.

Liger0, 2020-01-03:
Nevermind, I totally flashed it from 0 and it works now. The strange this is I didn't have to follow the steps 

1. Configure your ESP8266 for upload (USB to serial plugged in, GPIO0 and GPIO15 pulled low, RST pulled high)

2. Reboot the ESP8266 into run mode ((USB to serial removed , GPIO0 pulled high, GPIO15 pulled low, RST pulled high)

It deleted the previous datas (option delete all) and uploaded the new one without the first hardware connection and for the second I just did reset.

luc-github, 2020-01-03:
Great !

>Configure your ESP8266 for upload (USB to serial plugged in, GPIO0 and GPIO15 pulled low, RST pulled high)

Yes on nodemcu based boards this is not a need - 

>Reboot the ESP8266 into run mode ((USB to serial removed , GPIO0 pulled high, GPIO15 pulled low, RST pulled high)

and now esp core allows to boot without a restart after a flash if in nodemcu flash mode

Liger0, 2020-01-03:
Okay, I tried to connect the board via usb to the PC moving the jumper from left to right (this is needed to change the powering mode), and removing the wifi module.
Unfortunately the LCD just keeps showing "TMC communication error", and nothing happens whatever I send via arduino.

luc-github, 2020-01-03:
be sure of your serial configuration in Marlin, you must be able to communicate by usb port with FW

Liger0, 2020-01-03:
Well I think since the usb mode is limited and the hardware isn't powered I couldn't do anything anyway.

luc-github, 2020-01-03:
the limitation should be only Power related: heat and steppers movement. others things should work

Liger0, 2020-01-03:
Yeah it was only me having to use the other powering mode, now I can read from the machine if using pronterface (not using arduino) like in #81 .
My problems are exposed enough there I think, though I always am having random behaviours on the ESP3D web ui (not browser dependant, I always try multiple ones).

luc-github, 2020-01-03:
Issue is not in web ui I tested chrome and firefox without problem, issue is communication between printer and esp board

Liger0, 2020-01-03:
I'll close this and see if, now that it works with the updated firmware, I see any strange thing in that page again (except the file shown have wrong names as in the other ticket), I'll eventually come back.
Thanks. :)

closed 2020-01-03 by Liger0