luc-github/ESP3D-WEBUI #84: [FEATURE REQUEST] Only automatic scrolling when bar hits the bottom

Liger0, 2020-01-04:
**Is your feature request related to a problem? Please describe.**
Not a real issue, but it would make things easier.
The automatic scrolling is a cool feature but it could be improved in my opinion.
If you are reading earlier things, the commands will move continuously and make it hard to read a certain line. You can deselect automatic scrolling but I tend to forget and it is a bit annoying.

**Describe the solution you'd like**
Only engage automatic scrolling when the last line is totally visible/scrolling bar is at the bottom.
(Smart scrolling).
Thanks. :D

luc-github, 2020-01-04:
that would be a nice improvement indeed

luc-github, 2020-04-24:
seems not so easy windows buffer is limited to 300 lines so when buffer is full old lines as removed accordingly number of new lines then  new lines are added, which make every line in between shifted to the top 
the result is even with autoscroll off - there is some scrolling ...
Need to re-think the complete mechanism I think

luc-github, 2020-05-05:
ok I did a mitigationc in 3.0 to avoid text moving when buffer is full
if the scrolling is paused because scrollbar is moved up : the buffer is doubled to avoid to get a movement when buffer is reaching limit
then when scrollbar is put back to low - buffer is cleaned to normal limit

luc-github, 2020-05-23:
I think it will be only in 3.0 so I close issue

closed 2020-05-23 by luc-github