luc-github/ESP3D-WEBUI #85: [FEATURE REQUEST]Be able to autodiscover all ESP3D device and agregate them

luc-github, 2020-01-04:
One <s>Ring</s> UI to rule them all, One <s>Ring</s> UI to find them,
One <s>Ring</s> UI to bring them all and in the darkness bind them

A way to agregate all UI and control them like an esp3d farm, even sending jobs from one to another
This tool could also handle printers access for fab lab or a classroom  for example

It could be used also with usage of NFC devices - imagination is just the limit

This thread is more for a discussion  for the moment to see if there is a need or it is just a fake good idea ^_^

ETE-Design, 2020-09-27:
Great idea, that way it would be possible to control all printers using ESP3D in one GUI :-) Will you make an app for it to add all IP's, or one ESP3D Master and the rest Slaves? Would be even better to find a way to incooperate a slicer also so you can send the STL direct from Phone or whatever to the printers :-)

luc-github, 2020-09-28:
For the aggregation the idea is simply to use the  `MDNS.queryService("esp", "tcp");` API so any ESP can be master/slave - it is just a query so need to fine tune the management of this query, may be using https://github.com/luc-github/ESP3D/issues/421 and display result to webUI in the same way the scan network send information.

The offline slicer idea is already done by someone - but I think code is not public yet or won't be public, it is based on Cura engine (https://github.com/luc-github/ESP3D/issues/487#issuecomment-683441494)

The online slicer idea is pretty old but need a reliable online solution that provide online commands, and this need lot of test before choose the right one ( this one looks nice but never tested https://icesl.loria.fr/slicecrafter/, another one Astroprint is well known https://www.astroprint.com/online-stl-slicer-and-gcode-x3g-generator but not free and again need time to test and check how to integrate) - so my time bandwidth is the problem ...

luc-github, 2022-08-15:
the command to list all ESP3D / ESP3DLib / etc is now implemented https://github.com/luc-github/ESP3D/issues/803
Just need to define what UI would be convenient , the ESP450 generate a json file 
```
{
 "cmd": "450",
 "status": "ok",
 "data": [
  {
   "Hostname": "esp3d-test.local",
   "IP": "192.168.2.130",
   "port": "80",
   "TxT": [
    {
     "key": "version",
     "value": "3.0.0.a204"
    },
    {
     "key": "firmware",
     "value": "esp3d"
    }
   ]
  },
  {
   "Hostname": "esp3dgrbl.local",
   "IP": "192.168.2.200",
   "port": "80",
   "TxT": [
    {
     "key": "version",
     "value": "3.0.0.a203"
    }
   ]
  }
 ]
}

```