luc-github/ESP3D-WEBUI #86: [FEATURE REQUEST] E1 temp graphic

Liger0, 2020-01-04:
**Is your feature request related to a problem? Please describe.**
There is no graphics for the E1 temperature.

**Describe the solution you'd like**
Graph for E1 temp, there seems to be enough space.
Thanks!

luc-github, 2020-01-05:
It is actually a bug, for some reason E1 line does not show up in extruder graph
Long time I did not tested and seems something is broken - I will have a look

luc-github, 2020-01-05:
Fixed :
[image]

That was super old bug actually ... when moved E1 to preferences I forgot to update UI in apply function orz

closed 2020-01-05 by luc-github

Liger0, 2020-01-05:
The E1 graph doesn't seem very linear, the right side is blank most of the times and the curve is made of very large segments, like it being scanned less times than other ones.

luc-github, 2020-01-05:
There is no difference in calculation neither display process but color between E0 and E1

Liger0, 2020-01-05:
This is what I mean:
[image]
he E1 line keeps having that space on the right with the same offset.

luc-github, 2020-01-05:
Do you heat E1? 
My output is with Repetier FW, what is Marlin M105 output?

commit 9b71d007 (2020-01-05): Fixes #86