luc-github/ESP3D-WEBUI #87: [BUG] Axis coordinates don't show in the control tab

Liger0, 2020-01-05:
**Describe the bug**
Axis coordinates often don't show in the control tab.

**To Reproduce**
Steps to reproduce the behavior:
1. Activate the position check (I have it to 10 seconds)
2. It will show the coordinates and update 1 or 2 times
3. It will stop showing coordinates.
4. After this no matter what you do, it won't show coordinates again.

**Expected behavior**
Keep showing coordinates instead of blanking them

**ESP3D Firmware:  (please complete the following information)**
- Version: 2.1b39
- Web UI Version: 2.1b54

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version: 2.x bugfix as of 2 days ago

**Board used (please complete the following information):**
 - MCU:  esp8266 
 - Name: wemos d1 mini pro v1
 - Flash size: 16MB

**Browser used (please complete the following information):**
- Name: any
- Operating system: Windows 10

**Additional context**
The coordinates are reported correctly in the commands tab with detailed mod ticked.

luc-github, 2020-01-06:
Do you have screen shot?

Liger0, 2020-01-06:
Not a screenshot, I will to do later when I finish the long print I am doing. 
I can't do it now because always, mid print, the commands tab just keeps throwing
echo:busy: processing
multiple times and never connects to the printer again (same happens after a print is ended, it stops communicating and I have to restart the control box (the screen still shows the IP correctly).

I didn't change position. Just started a print.
I just tried to enable and disable the checkbox of the coordinates showing (leaving a minute between any click), but with no result so far.

luc-github, 2020-01-06:
Print from sd ? Or from usb?

Liger0, 2020-01-06:
SD

luc-github, 2020-01-06:
Busy: processing come from Marlin https://github.com/MarlinFirmware/Marlin/blob/9badd2b664463557d17449878b91ce84d5eec676/Marlin/src/core/language.h#L177
Seems it does not accept commands there nothing ep3d can do

Liger0, 2020-01-12:
Nothing so far. I guess it should be closed?

luc-github, 2020-01-15:
Ok as if Marlin does not allow any command when doing SD print - ESP3D cannot do anything
Anyway Marlin has auto temperature report to follow temperatures, so no need to poll  and it works with ESP3D- but I do not think it has same setting for position.

closed 2020-01-15 by luc-github