luc-github/ESP3D-WEBUI #89: Please add chinese languages,Thanks very much.

zrwd01, 2020-03-14:
**Describe the solution you'd like**
Please add chinese languages,Thanks very much.

**Additional context**
"cn”:“中文”,
"STA”:“客户端模式”,
"AP”:“AP模式”,
"BT”:“蓝牙”,
"Hold:0”:“已完成。准备恢复。”,
"Hold:1”:“进行中。重置将发出警报。”,
"Door:0”:“门已关闭。准备恢复。”,
"Door:1”:“机器停止了。门仍然开着。直到关门后才能恢复。”,
"Door:2”:“门打开。保持（或停车缩回）进行中。复位将发出警报。”,
"Door:3”:“门已关闭并正在恢复。从停车处恢复（如果适用）。重置将引发警报。”,
"ALARM:1”:“已触发硬极限。由于突然停止,机器位置可能丢失。强烈建议重新归位。”,
"ALARM:2”:“软限位警报。G代码运动目标超出了机器行程。保留了机器位置。可以安全地解除警报,请单击复位按钮。”,
"ALARM:3”:“运动时复位。由于突然停止,机器位置可能会丢失。强烈建议重新归位。”,
"ALARM:4”:“探测失败。在开始探测循环之前,探测未处于预期的初始状态。”,
"ALARM:5”:“探测失败。在G38.2和G38.4的编程行程内,测头未与工件接触。”,
"ALARM:6”:“回原点失败。复位了有效的回原点周期。”,
"ALARM:7”:“归位失败。在归位周期中安全门打开。”,
"ALARM:8”:“归位失败。拉出行程未能清除限位开关。尝试增加拉出设置或检查接线。”,
"ALARM:9”:“归位失败。无法在搜索距离内找到限位开关。请尝试增加最大行程,减小下拉距离或检查接线。”,
"error:1”:"G码字由一个字母和一个值组成。找不到字母。”,
"error:2”:“缺少预期的G代码字值或数值格式无效。”,
"error:3”:“未识别或不支持Grbl'$'系统命令。”,
"error:4”:“接收到的负值为期望的正值。”,
"error:5”:“归位循环失败。未通过设置启用归位。”,
"error:6”:“最小步进脉冲时间必须大于3usec。”,
"error:7”:"EEPROM读取失败。将受影响的EEPROM自动恢复为默认值。”,
"error:8”:“除非Grbl为IDLE,否则无法使用Grbl'$'命令。确保作业期间的平稳操作。”,
"error:9”:“在报警或慢进状态期间,G代码命令被锁定。”,
"error:10”:“在未启用归位的情况下也无法启用软限制。”,
"error:11”:“每行超出最大字符。未执行接收到的命令行。”,
"error:12”:"Grbl'$'设置值导致步速超过支持的最大值。”,
"error:13”:“检测到安全门已打开并且门状态已启动。”,
"error:14”:“构建信息或启动行超出EEPROM行长度限制。行未存储。”,
"error:15”:“点动目标超出了机器行程。点动命令已被忽略。”,
"error:16”:“点动命令没有'='或包含禁止的g代码。”,
"error:17”:“激光模式需要PWM输出。”,
"error:20”:“在块中发现不支持或无效的g代码命令。”,
"error:21”:“在块中找到的同一模态组中有多个g代码命令。”,
"error:22”:“进纸速度尚未设置或未定义。”,
"error:23”:“块中的G代码命令需要一个整数值。”,
"error:24”:“有多个g代码命令,需要在块中找到轴字。”,
"error:25”:“在块中找到重复的g码字。”,
"error:26”:“在g代码命令或需要它们的当前模态下,在程序段中找不到轴字。”,
"error:27”:“行号值无效。”,
"error:28”:"G代码命令缺少必需的值字。”,
"error:29”:“不支持G59.x工作坐标系。”,
"error:30”:"G53仅在G0和G1运动模式下允许。”,
"error:31”:“当没有命令或当前模态使用它们时,在块中找到轴字。”,
"error:32”:"G2和G3圆弧至少需要一个平面内轴字。”,
"error:33”:“运动命令目标无效。”,
"error:34”:“圆弧半径值无效。”,
"error:35”:"G2和G3圆弧至少需要一个平面内偏移字。”,
"error:36”:“在块中找到未使用的价值词。”,
"error:37”:"G43.1动态刀具长度补偿未分配给已组态的刀具长度轴。”,
"error:38”:“刀具编号大于最大支持值。”,
"error:60”:"SD无法挂载”,
“error:61”:"SD卡无法打开文件进行读取”,
“error:62”:"SD卡无法打开目录”,
“error:63”:“找不到SD卡目录”,
“error:64”:"SD卡文件为空”,
"error:70”:“蓝牙无法启动”,

luc-github, 2020-03-14:
好的謝謝

luc-github, 2020-03-15:
@zrwd01  did you checked the template here : https://github.com/luc-github/ESP3D-WEBUI/wiki/Translation-support

the translation you provided is limited to GRBL messages helper, not to webUI, is that normal ?

zrwd01, 2020-03-17:
I will try

luc-github, 2020-03-17:
@zrwd01 thank you - I have added the zh-cn framework with your translations
put the missing ones here again when you are ok I will update

zrwd01, 2020-03-19:
@luc-github 
[zh_CN.zip](https://github.com/luc-github/ESP3D-WEBUI/files/4355327/zh_CN.zip)
hi，I did some translations

luc-github, 2020-03-19:
now part of https://github.com/luc-github/ESP3D-WEBUI/tree/2.1
thank you ^_^

closed 2020-03-19 by luc-github