luc-github/ESP3D-WEBUI #92: [Question]Additional default settings

dchauran, 2020-03-24:
I have gotten this firmware working for the FLSUN QQ-S in combination with Marlin firmware. I would like to create a version of this customized for this setup, however I am not able to find a way to set some important defaults, such as firmware type, and to specify that bed heat and fan controls should be enabled.

The latter 2 look like I can attach a preferences.json to set them, but I would prefer the initial config for the FW to include these, and I don't see any way at all to specify a default firmware.

Is this possible?

dchauran, 2020-03-24:
Oops, accidentally posted under webui, not firmware.

closed 2020-03-24 by dchauran

luc-github, 2020-03-24:
@dchauran only FW target seting is in ESP3D FW, others settings are in ESP3D_WEBUI
you can set default values in preferences.json template which is in preferences.js  so they will be used if no preferences.json is found, but it won't be taken in account if there is already a preferences.json is already existing due to previous installation.

I use the FW target set to unknow to trigger the automatic launch of setup wizard, so if target FW is set, user need to manually launch wizard

May I know why you would hardcode default settings for this printer ? 
Setting FW target manually and  enable bed and fan control is not too painful for user, in worst case you can even share a preset preferences.json

dchauran, 2020-03-24:
The userbase for this printer is large, and has a varied skill level and diverse languages. I'm trying to make something that requires the least amount of documentation possible. I did end up bundling a configuration.json, but even that is more steps for a user to follow. It's not a show stopper or anything, but thought I would ask.

luc-github, 2020-03-24:
well there is something I wanted to do in 3.0 or 3.1  is to add a profile option in webUI wizard
the webUI would connect some website to download a full profile for ESP3D+WEBUI by just selecting the printer name in list box
This is actually what you looking for ?

dchauran, 2020-03-24:
That seems like it would suffice.

luc-github, 2020-03-24:
as it need web backend to store all configs - I was thinking to host the json file containing all info on github in a sub directory of ESP3D-WEBUI repository

dchauran, 2020-03-24:
That's actually a great idea, because it makes it easy for people to contribute a config for their printer.