luc-github/ESP3D-WEBUI #93: printer firmware upload

drunkscientist, 2020-03-25:
this is either a question or a feature request: im not sure i understand exactly which chip is being updated with the orange button on the esp3d tab of the webui. the header is "ESP3D Firmware Update" which leads me to assume it updates the esp8266 firmware, not the arduino. is there a way/would it be possible to update the printers firmware with the webUI?

luc-github, 2020-03-25:
'ESP3D Firmware Update' will update ESP3D Firmware ...
Printer Firmware update depend on printer board and printer firmware, currently ESP3D does not update printer FW.
Some printer boards allow to flash if FW is on SDcard some only by serial.

drunkscientist, 2020-03-26:
ok then definately a feature request, [since the 8266 is capable of programming an arduino](https://hackaday.com/2014/11/13/programming-an-arduino-over-wifi-with-the-esp8266/), but ill close this for now since i dont know if it should go here or under the standard esp3d page, and because i dont know if there would require additional wiring

closed 2020-03-26 by drunkscientist

jan-gerard, 2020-04-14:
I run Octoprint, which has a printer firmware flashing plugin. The trick is to reset the printer (Anet A6, button on LCD panel) and within 1 second press the FW upload button on the Octoprint web UI.