luc-github/ESP3D-WEBUI #94: [DISCUSSION]New UI 3.X

luc-github, 2020-03-30:
Now Framework is defined and components almost defined [here](https://github.com/luc-github/ESP3D-WEBUI/issues/48#issuecomment-602223316), and wizard process is under definition [here](https://github.com/luc-github/ESP3D-WEBUI/issues/55#issuecomment-605603424)
It is time to discuss UI itself, to specify it's look and feel and behavior

Here a start :
* menu items should be limited to minimum, and should be collapsed to list menu if mobile device 
   - ESP3D -> an about box listing informations about current settings / browser / versions, links to Github and allowing web update feature for FW and UI
   - Dasboard -> everythin g related to machine (defined later)
   - Settings -> every settings in one place : ESP3D wifi / features settings, Machine settings / webUI preferences 

   - Login -> if authentication is enabled, it will allow to log out and change user password 
  - a camera switch if camera feature is enabled (on / off / overlay / embedded)
  - a fast access to Terminal command, not always need one but when need better to be easily accessible
* Notification area - always visible, it should display important output informations
   - Status of current job ongoing if possible
  - Temperatures for 3D printer, including DHT if any, option to display graph
  -  Machine positions 
 * if dashboard is selected 
   - Macro buttons ribbon, which should be possible to collapse to a simple emergency stop button
   - Jog controls for all axis including extruders
   - Specific input controls according machine: 
        * Feedrate / flow / fan control / temperature controls for 3D printers according settings
        * Mist / flood / spindle / probe controls for GRBL according settings
   - a File system panel - this one whould be collapsed by default as it is not an permanent usage , it would show all available filesytems available, ESP3D/all SD/Udisk if any etc...
  and allow to manage files and start a job when possible

* if settings is selected
  - a wizard button to get slimplified setup 
  - ESP3D wifi settings and feature settings (with import/export button and esp restart button)
  - Machine settings (with import/export button)
  - WebUI preferences settings (with import/export button)
  
The idea is to get most of important output first, then input according frequency of usage
e.g. no need to permanetly show settings neither files systems

Feedback suggestions are welcome

luc-github, 2020-04-04:
About page is done
it display all information about ESP3D and some about UI
there are 2 buttons to upload new FW and new UI files
Connection check is implemented with auto reconnection

here some screenshots

Desktop UI
[image]

Mobile UI
[image]

Next week I will work on Settings Page

Current index.html.gz is under dist directory for early test : https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/dist/index.html.gz

luc-github, 2020-04-13:
settings is ongoing - still on ESP3D settings this week because I needed to setup some workflow
and new UI, and list all translated string.

[image]

[image]

Setting are grouped by section
Only valid modified setting will have a button so save setting
Error message will be under faulty setting
The save feedback is real one it appear only after saving and disapear after refresh
I removed the reset button to simplify UI because the refresh button actually do the same

So result is a simplified UI that will be consistent accross all settings

I still need to implement the `Flag` settings and will add the import/export functions

luc-github, 2020-04-14:
setting has now the scan dialog :
[image]

luc-github, 2020-04-14:
Export settings is now implemented 
[image]

The button directly save the file to PC, it seems not possible to do a 'SaveAs' to select destination - but at least settings are exported in JSON file from web page, next step is import this file 😸

luc-github, 2020-04-15:
[image]

Import and board restart are  now implemented - next step will be UI preferences settings for current UI settings  then fw target settings

labotecno, 2020-04-21:
nice work ! 
do you have something upload?

luc-github, 2020-04-22:
everything is here : https://github.com/luc-github/ESP3D-WEBUI/tree/3.0
I build time to time index.html.gz here : https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/printer

But you need to have ESP3D 3.0 on ESP to makeit work

or you can use the commands `npm install` then `npm run dev` (check the [readme](https://github.com/luc-github/ESP3D-WEBUI/tree/3.0#setup-development-tools) )and you have a local version and no need ESP board to test

luc-github, 2020-04-23:
UI settings have some check boxes and there is no way to use default checkbox and add visual feedback on it -  so I have created a custom one 
[image]
if setting is modifed and not saved
[image]

when settings are saved :

[image]

like this rendering should be consistent

TheZoker, 2020-04-23:
Wow looks very nice!! Only one suggestions: Add some right padding to the icons on the buttons, so there is some room between the icons and the button text 🙂

luc-github, 2020-04-23:
for the button padding text  you are correct need modifications was not enough and not same everywhere actually 
now it is in class css I have put 20px, may be cam decrease a little :
[image]

for the setting selection well I tried tabs before and I was not happy with the rendering - so I used button like but not to spaced  - please note on mobile UI there is no text 

[image]

If you have better rendering feel free to try - this bar is in `ESP3D-WEBUI\src\components\settings\selector.js`

TheZoker, 2020-04-23:
How about something like the button group from bootstrap?
[image]

luc-github, 2020-04-24:
yes tried also when looking for a solution and not really find nice  for this usage too much stick together, so looks weird full pc page to have so tiny control , and for touch device I think it is better control has clear visible separation when they are close to each others

IMHO

luc-github, 2020-04-24:
Languages pack are now supported
[image]
When selecting the UI check if file is present on filesystem    
and raise error if not  

[image]
The build script compress to gzip language pack and put then in dist directory     

[image]
the test server use the ungzip files to allow modification easily   
[image]

Currently the language list is hard coded in page, but last item allow `custom` file so any file named custom.json.gz will be taken - the check of presence is done at each change but nothing prevent to save, if package is not found at start english is used, if lagnguage pack is incomplete, english also is used.

luc-github, 2020-04-25:
Next steps roadmap : 
1 - Machine settings
 * repetier settings
 * marlin settings
 * smoothieware settings
with import / export feature as other settings

2 - Terminal commands
need to add a control that collapse/expand the terminal command to send and monitor commands

3 - The jog control -TBD - I am still wondering if I improve current UI based on pronter face or define/use a new one 

Feedback are highly welcome

4 - File upload / monitoring - based on different possible SD , USB Disk/ etc.. TBD 

5 - Specific panel according target Printer (temperature/extruder/ status) / CNC (probe/ status)

luc-github, 2020-05-02:
Printer CNC settings tooks more than I expected but it is done
This feature bring also the split between printer and CNC code and translation

I have tried to make consistent UI between FW which was a big challenge 
* Smoothieware has 2 configurations sources:
 - config.txt.  a text file having lot of commented lines and text
 -  and config-override, a gcode file having a commented lines and text
* Marlin has a Use EPPROM using GCODE commands which also have some comments when display contents
* Repetier has an EPPROM using 1 gcode addressing  specific locations  in EEPROM with embedded help for units on some commands

Marlin PC view:    
[image]

Marlin Mobile view:   
[image]

Repetier  PC view:   
[image]

Repetier Mobile view:    
[image]

Smoothieware PC view :   
[image]

Smoothieware Mobile view:   
[image]

Smoothieware  config-override PC view :  
[image]

Smoothieware config-override Mobile view:   
[image]

GRBL PC view:   
[image]

Grbl Mobile view :
[image]

Label using text can be long and with translation in some language it can be even longer - so  in mobile view text font is decreseased to keep space available for input

Commented lines in smoothieware are a lot and useless because there is not "easy" way to uncomment a line currently so I decided to remove them, I kept only the comment at the end of each active setting and put then as settting header, same for comments in config-override

In Marlin I use comment as setting headers also

Repetier has no comments but help for units  - so when put them at the end of input

Grbl label are not always meaning full so the header is a `translation` of it like in 2.1 

Comment and Help are hidden in mobile view to keep view clear
The save button do a `M500` command when reset send 'reset' to smoothieware to reload the new configuration at start

all setting can be imported and exported

To check the value enterred is correct I have added some basic test according FW constraint and tried to use some regex for validation - this part need to be improved but should be ok for the moment

all http commands answer using GCODE are catched by websocket  

Now printer index.htmlgz is 46.3 KB when grbl is 45.6 KB
[image]

languages pack are generated and compressed in each target directory
[image]

I have splitted what is specific printer and what is specific grbl 

I did not tested on real system yet - only on dev mode which simulate esp3d + printer responses so some adjustment may necessary on live system, I will do it after the next step implementation

Next step :  Terminal commands

luc-github, 2020-05-05:
New terminal commands paenl is now implemented, it is hidden by default:   
[image]

Chrome browser:
[image]

Firefox browser:
[image]

it should work around https://github.com/luc-github/ESP3D-WEBUI/issues/84, I have set a custom scrollbar for UI consistency

luc-github, 2020-05-05:
Before working on Jog panel I will do some test for notification panel
now there is a an automatic generation of temperature for test I can see how to display processed  values in notifcation area - may be using the  storeon api of Preact 
also this is may be a better way to update terminal content  than what I am doing right now

luc-github, 2020-05-06:
Ok POC using storeon for temperatures and positions is pretty small so implement (only the mecanic) with small filter for the data detection but increase the final build from 47.4KB to 50.6KB, adding the terminal output in this mechanism has low impact as kept the final size to 50.6 KB so will be use when working on point (5)

luc-github, 2020-05-06:
Now on point 3 - **The jog control**
- I am now `upgrading` the current UI  SVG based for 3D printer to give better look and feel adding pointer cursor, pressed button state and tooltips when no text   
[image]

This should be PC view, for mobile view I will use squared buttons based UI 

For CNC should be squared buttons based UI also as axis can vary from 1 to 6

I am not yet fixed for buttons based layout but I am thinking to have
*  3 buttons for each axis (+ - home) for 3D printer
*  4 buttons for each axis (+ - home zero) for CNC

Distance will be defined by radio buttons 0.1 1 10 100 

of course need 2 inputs : XY Feed rate and Z feed rate for 3D printer
for CNC the PC display should show up to 6 axis when mobile view will have X, Y, (Z, A, B,C) with selction box

TBD this week

luc-github, 2020-05-09:
So I do the base UI for Printer - CNC will follow once all is done for printer (jog UI + API)
I have move speed = feedrate to Jog panel because it is related, I do not know if it is a bad idea ,but let's give a try ...
Here the mobile view, I tried to keep as simplest as possible touch oriented :   
[image]

Here the PC view which is more fancy and mouse oriented:
[image], I am still considering for pc to put all controls in column on side of SVG instead of under as responsive line

I will implement the API mechanicsm + preferences settings
Feel free to feedback

ETE-Design, 2020-05-10:
@luc-github Are you only running this in emulate mode? Cause I have tryed it on my printer, when I click at Settings i get an Error message: Cannot connect with board... And then just a blank screen, so I can't change anything to let it connect...

luc-github, 2020-05-10:
yes only emulate mode for the moment will do some test next week with some system

ETE-Design, 2020-05-10:
Okey, cause when trying to use it on a printer you will get an error on the Setting Page, and then have no possibility to change settings to let it connect to the printer, so you have to make some changes so if it does not connect it will still show the setting page to let you make the changes needed to connect...

luc-github, 2020-05-10:
I have only worked on UI and emulator for the moment - emulator is the ideal case - so I am sure some tuning will be necessary when used in real case for UI and FW

ETE-Design, 2020-05-10:
It is, it dosen't know what to do if the printer is not connected ;-)

luc-github, 2020-05-11:
@ETE-Design fixed
I have added a timeout in UI if printer does not answer allowing to get UI back after 15sec
This mechanism should cover any communication issue between printer and ESP

I have also added the send command echo in terminal with a `>` prefix + and added some others minor fixes

About the printer not answering - this was due to web command do not add "\n" so printer was waiting for it - small thing but important...
[image]
I am sure the verbose mode will need some tuning to dfine what need to display and what need to be hide

I did my quick tests on due + esp8266 and so far so good to get printer config,  
[image]

You should be able to use the terminal to communicate with printer now

I do not plan to test more for the moment so I will work implementing Jog commands this week

ETE-Design, 2020-05-11:
@luc-github I Have an ESP8266 and ESP32-Cam... On the ESP8266 it work as expected, but it's another case on the ESP32-Cam, if you have an ESP32-Cam it would be great if you could test it on that also :-)

luc-github, 2020-05-11:
@ETE-Design please open issue if you see something describing issue 

 I cannot try all plateforms/configurations each time I implement a new part of UI 
I will do some test time to time if implementation need backend to validate but most of the time I am using the emulator or it will take edge before something is done

of course I will do a full validation on all board I have at the end of above development plan but I won't continuously try on every board/configuration until the UI is finalized

luc-github, 2020-05-12:
Jog panel for 3D printer is now finalized 
PC view:
[image]

Mobile view:   
[image]

I will do GRBL now using Printer mobile view as a base, no fancy UI, should be done this week, the main difference is GRBL has 1 to 6 axis when 3D printer has always 3

luc-github, 2020-05-14:
I have changed radio button to press button in printer mobile view for ease of use
[image]

GRBL Jog is mainly done - some override control will be added later
GRBL PC view:    

[image]

GRBL Mobile view:
[image]

Next step is file selection panel, ESP FS, SD, USB, etc...
I plan to make it displayed using a  toggle like for terminal as usage is limited to upload files and start print job, a drop box to select which FS to display should cover all possibilities

luc-github, 2020-05-24:
backbone for Filesystem UI is now done and ESP FS is implemented - still need to add the SD, USB, etc.. but everything is almost prepared 
PC View:   
[image]
[image]

Mobile View:   
[image]

preferences UI start to have some content 😸  :
[image]

I need to review/cleanup webUI ESP FS implementation, still have couple of TODO and clean debug code then I will go back to ESP3D 3.0 FW to fix some glitches I have noticed with SPIFFS and ESP32, so need to test also ESP8266 and implement port Serial Upload from 2.1 which is not yet done 😭 

Once done, I will go back to WebUI for Serial SD / TFT SD/ TFT USB and Direct SD for Marlin / Repetier and Smoothieware ... 😼 🧑‍🏭 

Step by step ^_^

luc-github, 2020-05-25:
Download file monitor is now implemented 
[image]

So ping will be desactivated when doing download and any other action will be also blocked until download is done 😸

luc-github, 2020-06-12:
Quick update - I spend some time to do some long run and some stress tests and fix them on WebUI,  
I have started the SD Serial addition : 
[image]
The devt was slow down due to external events but now I should be back to focus on webUI

Hope to do some progress quickly

luc-github, 2020-06-16:
I have done the Repetier / Marlin / BTT TFT SD / BTT TFT USB / Smoothieware for serial SD, upload is not implemented in UI neither FW - I still have the GRBL_ESP32 and ESP3DLib to do but FW is not done neither - so I will hold this part until FW is ready

I will continue by notification area / macro support / login support / Camera

luc-github, 2020-06-20:
For notifications I plan to display :
* machine name (only in desktop mode)
* connection status (connected / connecting)
* sensor output 
* temperatures
* positions
* feedrate %
* flowrate %
* fan %

clicking on any one will display corresponding control : temperatures / feedrate / flowrate / fan

Which means jog may become visible only when position is clicked
also  extruder control for extrusion should appear in position ? or when select extruder itself (TBD)

So for printer only motor of and emergency stop buttons will appear under terminal and files button
making UI simpler - only displaying what is used

luc-github, 2020-06-22:
Because recurent output like `wait` can easily make output noisy I have added same feature as console.log output
before ouput can be 
[image]
 Now I aggregate same successive output - which decrease the necessary lines terminal and give more clarity 
[image]

not verbose mode is still clean
[image]

luc-github, 2020-06-23:
herePOC UI for 3d printer
[image]

Still need to add flow / feed / fan / sensor and connection status

Also need to add hot spot to show the correcponding control (temp / jog / flow / feed / fan )

I have tryied to make it simple - easy to read (I hope) -
When heating label is red and show target temperature - I am not super happy for target temperature display, but I am dry,  so if any better idea : feedback /suggestions  are welcome, same for bed icon done by me 😼

DivingDuck, 2020-06-24:
Nice work. An other way to display actual/target temps is using gauges maybe upside beside of z+/z-.

A whish, is it possible to support more than one heated bed for bigger printers?

luc-github, 2020-06-24:
@DivingDuck, interresting - I will check gauge for PC view 

About multiple bed support, why not - I remember I read something about a printer having, one for central for small objects and one for external area,  it but I have no idea how to monitor /control  them.
Marlin / repetier seems only supporting one - do you have such configuration ? can you describe how bed are controled in such configuration?

DivingDuck, 2020-06-24:
Yes, I have :)
It's based on Smoothieware, where I had add an additional temp module that is controllable via g-code and console commands. I modified in addition Pronterface to have control over the two beds. In Cura and Slic3r I defined different printer profiles to make slicing with two beds manageable as I want to use different combinations like bed 1 and/or bed 2 or both with two separate glass plates or one big plate depending what is to do.

[image]

Pronterface is reading the temperatures from M105, where I define the second bed as B1. Like:
[RECV: ok T:29.9 /20.0 @18 T1:28.8 /20.0 @255 B:29.7 /30.0 @255 B1:29.4 /30.0 @255]

Next step will be to think about a more automated slicing where the slicer know about when to turn on a bed,

luc-github, 2020-06-24:
Ok `B1:29.4 /30.0` is super easy to monitor as I just need to add it

What Gcode heat the Bed?  `M140 S80 T1` `M190 S80 T1` ?

DivingDuck, 2020-06-24:
That is reprap, correct? I whish it where so easy :)
 I use custom M-codes as it wasn't supported that way with smoothieware when I implemented this. Guess, better you let the user set up an individual M-code. I use @set_temp bed2 30.0 via command line and for M-Code M1004 S80 (like M140) and M1009 S80 (like M190) for the second bed.

luc-github, 2020-06-24:
ho that means more custom control to setup in preferences

luc-github, 2020-06-24:
> M-Code M1004 S80 (like M140) and M1009 S80 (like M190) for the second bed.
@DivingDuck in that case why not setup the additional bed like an additional extruder ?

DivingDuck, 2020-06-24:
Yes, as there is not a real standard right now. https://reprap.org/wiki/G-code#M140:_Set_Bed_Temperature_.28Fast.29.

To your question to set it up as additional extruder, I was thinking so too, but I plan to use a tool changing system in the next evolution and then it will become a bit confusion what is a bed or a real extruder or what ever else. That's why I took this route. 

You remember me to ask the smoothie guys if it is possible to integrate the way, RepRap is doing it for M140 - but I guess, they don't like it...

luc-github, 2020-06-24:
here gauge done using light svg code :
[image]

DivingDuck, 2020-06-24:
looks nice but took a lot of space if you want to show more temperatures. Maybe forming a pillar?

ETE-Design, 2020-06-24:
@DivingDuck Nice with the 2 bed options... I'm rebuilding my CR10 S5 with a silicone bed with 2 zones, over 2 SSR... Though I run Marlin, do you know if there is an solution for control them individual via G-Code? Optimal would be that Cura Will turn on 1 / 2 zones depending on object size :-D

luc-github, 2020-06-25:
@DivingDuck yes I know, and I am not good a design - that is why I stick to simple UI ^_^

DivingDuck, 2020-06-25:
@luc-github, don't understand this wrong,your gui is beautiful. This is always also a matter of personal taste. I like the way you are doing this and the way you involve users to their thoughts about your implementation. I hope that there will be more people take the chance to comment your work to get it even better.
Yes, I build this kind of printers myself. It have two beds with 320x320, 500 in hight with 600W each.  It's based on Scott_3D HyperCube Evolution design.

@ETE-Design, I don't have a solution for Marlin, but you can think about the mentioned idea from @luc-github (as an additional extruder). This is one way I know that other peoples and companies had implement those solutions. Or use a RepRap firmware. I choose Smoothieware in the past because I can simply define as much temperature modules as I need in a configuration file w/o the need of compilation-orgies.

luc-github, 2020-06-25:
@DivingDuck thank you for your support - always nice to hear ^_^
What about the 2 area heat bed itself ? did you built yourself ? Currently I only found 500x500 silicon heat bed in one heat area, but none with 2 areas

ETE-Design, 2020-06-25:
@luc-github I have a 508x508 silicone heated bed in 2 zones, so it is possible... Found mine on AliExpress... But also have a contact in China so they can custom make em for my needs... But forgot to tell then to add an extra Thermistor to the 2 zone I have right now, so have to add it myself...

DivingDuck, 2020-06-25:
I have a different design approach. I use two separate 320x320 plates with two silicone heater. It looks like:
[image]

An other solution for such a big plate would be a custom silicone mate with two separate zones (sometimes I saw these at aliexpress) , but I do not prefer this for my builds. They will consume a lot of energy if you need high temperature and mostly it makes no sense to heat up a huge bed for smaller prints. In addition you will fight against warping plates if you do not heat them evenly. My guess is, that you will bend such a big plate over time if it isn't thick enough and you heat up only a section of the plate.

luc-github, 2020-06-25:
@ETE-Design thank you - do you have the shop name of the one you bough in aliexpress ? - seems I did not search well

@DivingDuck  the 2 area solution I think I saw on a finish product was: 1  central area for small object (20cm X 20 cm) and second  area is all around the first area, like frame  with 20cm heigh for a total bed  40cm x 40cm  - for whole bed - so in that case bot area need to heat

[image]

Yeah I agree even heat is no easy ti get on big area

ETE-Design, 2020-06-25:
@luc-github https://a.aliexpress.com/_d71YHqM

luc-github, 2020-06-25:
@ETE-Design thank you ^_^
Did dou sticked this on some aluminium plate ? is it working well ?

ETE-Design, 2020-06-25:
@luc-github Yes, put it direct on the aluminium plate, but as standard it only have one Thermistor, so need to add a 2'nd... Also have put in a PEI Steel plate to minimize weight... But haven't tryed it out yet cause still waiting for parts from China :-(

luc-github, 2020-06-25:
@ETE-Design ok, noted - thank you I will see what I can use to stick the silicone - for the moment I have nothing but a 550x550x6mm glass  so it is flat  but it is heavy and not heated, ^_^ I need to improve that

ETE-Design, 2020-06-25:
@luc-github I talk a Lot with the owner of the shop, so I'm sure on he can make an 550x550 if I ask him, also with screw holes where It's needed... There is already 3M glue on it, so No need to figure out how to fasten it...

luc-github, 2020-06-25:
@ETE-Design ho he can do custom size - that is good to know, actually I planned to use standard size 500 X500 - not custom one 
Ok noted - thank you

[40 more replies]