luc-github/ESP3D-WEBUI #95: Error when trying to upload any ".g" / ".gco" file

DenisG1994, 2020-04-17:
**Describe the bug**
Anytime I try to upload a file (g;G;gco;GCO;gcode;GCODE) via the "SD files" panel I get the error attached (Error 0 : )

**To Reproduce**
I tried several times both reflashing my esp8266 (Wemos D1 mini board) and reformatting the SD card of the printer (Anycubic I3 mega) but with any success. 
I'm using those versions for the board UI: 2.1b20 / FW: 2.1.0.b39 and the 2.5.2 for the Arduino boards. 
The esp can connect correctly to the printer, it can check and change the axis position and the temperatures. If I put manually a ".gco" file from a pc to the SD card it even see it and also print it.

**Expected behavior**
Expect to upload a ".gco" file

**Screenshots**
[image]

**ESP3D Firmware:  (please complete the following information)**
- Version:2.1.0.b39
- Web UI Version:2.1b20
- Wifi mode:tried both AP and CS (now in CS)
- No changes in FW

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version Stock 1.1.0 (hope I don't need to update cause of power loss recovery)

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name:Wemos D1 mini
 - Flash size: 4M (2M SPIFFS)

**Browser used (please complete the following information):**
- Name: tried Safari, Edge, Chrome
- Version  Latest release 
- Operating system: Mac OS10.15.3, WIN 10

**Additional context**

luc-github, 2020-04-17:
it means board reset due to power supply stability : https://github.com/luc-github/ESP3D/issues/417

DenisG1994, 2020-04-17:
I've read the other issue and tried those methods:
-Powering the board via the usb cable from my pc while connected to the printer (GND in common),
-Powering the board via a 12V -> 5V 1A voltage regulator connected directly to the PSU,
-Powering the board via a "desk voltage generator" both with 3.3V and 5V (dedicated pin on the board) the board never draw more than 0.09A,
-Powering the board via a "desk voltage generator" both with 3.3V and 5V (dedicated pin on the board) with a 470μF.
None of the previous test solved the problem... Don't know if it make any difference but I'm using a logic LevelConverter to allow communication between the printer and the ESP8266.
Uploading the test.gco present in the other issue fail as always...
Thank you for your assistance and your quick reply.

luc-github, 2020-04-17:
I do not use logic LevelConverter  so I cannot say if it is just for test you can try without
I know ESP8266 is very sentive for this

DenisG1994, 2020-04-17:
I will post all what I've done to install ESP3D, could you please tell me if there is something wrong?
First I have verified the correct flash size using the Example file provided, after that I've flashed the "esp3d.ino" file onto the ESP with the setting in the attachment (without making any change to the config.h or any other tab).
[image]
After a reset I have uploaded the 3 files present in the folder "data" inside "esp3d" to the ESP web page and have configured my printer and connection settings.
In the dashboard I can do all I want except to upload files to the SD card
I've found that I can upload files to the filesystem folder without any problem like in the picture.
[image]
Here are two photos of my wiring setup:
[image]
[image]
I've also tried another ESP board to make sure it's not defective.
Let me know if there is any other kind of test that I can do to make it work for transferring files via WI-FI. Thank you.

luc-github, 2020-04-17:
installation is correct and as said looks like power supply problem due to heavy need for upload

try this file in comment https://github.com/luc-github/ESP3D/issues/417#issuecomment-591975609
it is small and should work

DenisG1994, 2020-04-18:
Tried and retried to upload any kind of file of any dimension... Here are the screenshots.
Check that the trigorilla and the ESP communicate properly:
[image]
I can see files putted on the SD from my PC:
[image]
Choosing the file to upload:
[image]
The loading bar jump to 100% and then give the same error as always:
[image]
The SD panel look like that after that:
[image]

luc-github, 2020-04-18:
did you renamed the `test.gco.txt` to `test.cgo` ? I see you try `test` but you already have a directory 
 `test` so it won't work

luc-github, 2020-04-18:
can you do also a M115 and share the output ?

DenisG1994, 2020-04-18:
Yes I renamed it, for safety I added the ".gco" also to the name of the file...
[Video.mov.zip](https://github.com/luc-github/ESP3D-WEBUI/files/4496548/Video.mov.zip)
Sorry for the video quality but I had to do it to reduce size.
Here is the output of the M115 command:
M115
FIRMWARE_NAME:Marlin 1.1.0-RC8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

luc-github, 2020-04-18:
also have a test of this :
in terminal command 
type 
`M110 N0`
expected answer is `ok`
type `N1 M28 /mycode2.gco*121`
expected answer is
```
echo:Now fresh file: /mycode2.gco
Writing to file: mycode2.gco
```
type `N2 M117 Hi man*101`
expected answer is `ok`

type `N3 G1 X20 F1000*54`
expected answer is `ok`

type `N4 M29*28`
expected answer is 
```
Done saving file.
ok
```

let me know what you get

DenisG1994, 2020-04-18:
Here are the results:
[image]
I now have a file on the SD card named `MYCODE2.GCO`

luc-github, 2020-04-18:
so upload process is working

luc-github, 2020-04-18:
do a file named mygcode3.gco
with same content and upload it 
```
M117 Hi man
G1 X20 F1000
```

DenisG1994, 2020-04-18:
I got this error:
```
Error:Line Number is not Last Line Number+1, Last Line: 3
Resend: 4
```
Tried also with N4 and got this:
```
Error:checksum mismatch, Last Line: 3
Resend: 4
```

luc-github, 2020-04-18:
ho sorry 
keep same name

DenisG1994, 2020-04-18:
the one from my pc or the one generated from the web page?
Here are both the one I'm trying to upload and the other one generated via the ESP.
[test.gco.txt](https://github.com/luc-github/ESP3D-WEBUI/files/4496654/test.gco.txt)

[MYCODE2.GCO.txt](https://github.com/luc-github/ESP3D-WEBUI/files/4496653/MYCODE2.GCO.txt)

luc-github, 2020-04-18:
hmm 
on my side I get `N1 M28 /mycode3.gco*120` but you have `N1 M28 /mycode3.gco*121`
that is weird

DenisG1994, 2020-04-18:
Sorry, it was my fault, I was copy/pasting like before...
With that it worked:
```
M110 N0
N1 M28 /mycode3.gco*120
echo:Now fresh file: /mycode3.gco
Writing to file: /mycode3.gco
N2 M117 Hi man*101
N3 G1 X20 F1000*54
N4 M29*28
Done saving file.
```

luc-github, 2020-04-18:
looks like it cannot create file

luc-github, 2020-04-18:
can you do an M21 and M20 before upload

DenisG1994, 2020-04-18:
Sure, here is the log from the Arduino serial:
```
M20
M21
N1 M29*25
N8 M29*16
M21
M110 N0
M117 Uploading...
N1 M28 /test.gco*68
M117 Upload failed
N2 M29*26
N9 M29*17
M30 /test.gco
M117 SD upload failed
M117 Upload failed
N3 M29*27
N10 M29*41
M30 /test.gco
M117 SD upload failed
```

By the way, same error on the web UI

luc-github, 2020-04-18:
I am working on some output improvement when upload happen to follow the process and see what is wrong more easily

DenisG1994, 2020-04-18:
Ok, update me when it release so we can find a solution :)

luc-github, 2020-04-18:
@DenisG1994 you can give a try to https://github.com/luc-github/ESP3D/tree/2.1-devt
also use the index.html.gz present in data directory it is the latest one so we should be sync
I was able to upload all files 1KB, 300KB and even 3MB (which took a while....) without issue

DenisG1994, 2020-04-18:
Do you mean the 2.1.1 version? Cause I yet have installed the 2.1

luc-github, 2020-04-18:
yes

DenisG1994, 2020-04-18:
So... I've tried both the 2.1.1 and the 2.1-devt
With the 2.1.1 the error given is the same as the 2.1

With the 2.1-devt I got this one instead:
[image]
And the terminal fill itself with those lines:
```
M20
Begin file list
MYCODE3.GCO
BASE.GCO
MYCODE2.GCO
End file list
ok
N1 M29*25
ok
M21
echo:SD card ok
ok
M110 N0
ok
M117 Uploading...
ok
?	ýÿ?<ÿ?üþ?Èq @ä@Øüþ?ýþ?À¬!@
```
(The M20 was launched by me by hitting the refresh button)

luc-github, 2020-04-18:
the output is incomplete but file creation is failling

luc-github, 2020-04-18:
```
N1 M29*25
ok
M21
echo:SD card ok
ok
M110 N0
ok
M117 Uploading...
N1 M28 /mycode3.gco*120
ok
echo:Now fresh file: /mycode3.gco
Writing to file: mycode3.gco
ok
?"Ìüÿ?«!@N2 M117 Hi man*101
ok
N3 G1 X20 F1000*54
ok
N4 M29*28
Done saving file.
ok
M117 SD upload done
ok
Ok
@Xðþ?_s¦P@_s¦hþþ?ðþ?hþþ?Pþ?hþþ?´ÞM20
Begin file list
CUBE.GCO 112825
MYCODE2.GCO 27
MYCODE3.GCO 27
TEST.GCO 27
TESTCUBE.GCO 350777
FLEXHA.GCO 2958573
End file list
ok
```

[image]

[image]

DenisG1994, 2020-04-18:
Here it is, sorry for the copy/paste error
```
N1 M29*25
ok
M21
echo:SD card ok
ok
M110 N0
ok
M117 Uploading...
ok
?	ýÿ?:ÿ?üþ?Èq @ä@Øüþ?ýþ?À¬!@:N1 M28 /test.gco*68
Error:Line Number is not Last Line Number+1, Last Line: 1
Resend: 2
ok
M117 Upload failed
ok
?Ìÿ?üþ?¡!@BN@Xðþ?p>p @¬þþ?:N2 M29*26
ok
M30 /test.gco
Deletion failed, File: test.gco.ok
M117 SD upload failed
ok
```
[image]
[image]

luc-github, 2020-04-18:
obviously there is noise on your wires , your communications has garbage
try to secure your rx/tx wires may be your logic converter bring this noise

DenisG1994, 2020-04-18:
I've tried to change wires (shorter/thicker...) no results... I've connected an oscilloscope to check any data loss during TX or RX but still nothing... I was wondering if the fact that my printer uses a baud of 250000 could influence all the TX/RX thing... cause in the Arduino IDE I can't choose 250000 for my ESP chip...

luc-github, 2020-04-18:
i am using 115200

luc-github, 2020-04-19:
@DenisG1994  my  mistake :sweat: the garbage seems coming from a typo in my debug output 
I am working on additionnal debug output to add the debug messge now on top of just echo input/output

DenisG1994, 2020-04-19:
Ok, was wondering in fact why the garbage thing was always the same and never different from previous test... do you want me to test the current 2.1-devt?

luc-github, 2020-04-19:
yes looks like `M110 N0` is not take in account  to reset the line numbering
but should work if you restart printer as no previous commands

DenisG1994, 2020-04-19:
I've tried this after a reboot... Like you said `M110 N0` it's "ignored"
```
N1 M29*25
N1 M29*25
ok
ok
M110 N0
M110 N0
ok
ok
N1 M29*25
N1 M29*25
Error:Line Number is not Last Line Number+1, Last Line: 1
Resend: 2
Error:Line Number is not Last Line Number+1, Last Line: 1
Resend: 2
ok
ok
```

luc-github, 2020-04-19:
can you give me the M115 output ?

DenisG1994, 2020-04-19:
Here it is:
```
M115
M115
FIRMWARE_NAME:Marlin 1.1.0-RC8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
ok
FIRMWARE_NAME:Marlin 1.1.0-RC8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
ok
```
By the way, every command I send is sent twice.

luc-github, 2020-04-19:
yes the current FW echo all commands in webUI 
according to source code M110 N0 should work so I am not sure what would be the problem

I have just updated 2.1-devt - please give a try - debug is enabled and verbose (very)

luc-github, 2020-04-19:
I think I found the bug ... in Marlin 1.1.0-RC8 it has beed fixed later 
the M110 N0 do not change the last number used just the current 

```
/**
 * M110: Set Current Line Number
 */
inline void gcode_M110() {
  if (code_seen('N')) gcode_N = code_value_long();
}
```
in latest git 1.1-X
```
/**
 * M110: Set Current Line Number
 */
inline void gcode_M110() {
  if (parser.seenval('N')) gcode_LastN = parser.value_long();
}
````

do you have the source of your current FW ? - you mentioned you do not want to update it

DenisG1994, 2020-04-19:
Anycubic share only the `.hex` file available from their website here https://www.anycubic.com/blogs/news/all-you-need-to-know-about-i3-mega-m-s
The file is `i3 Mega  Firmware V1.1.5` yesterday I uploaded it to the printer but via M115 it seem to non have changed...

luc-github, 2020-04-19:
well if M110 is not working the only hope is upload may not need check sum

DenisG1994, 2020-04-19:
Is there a way to disable it?

luc-github, 2020-04-19:
try this before
in terminal command
type
`M29`
expected answer is `ok`
type `M28 /mycode2.gco`
expected answer is 
```
echo:Now fresh file: /mycode2.gco
Writing to file: mycode2.gco
```
type `M117 Hi man`
expected answer is `ok`

type `G1 X20 F1000`
expected answer is `ok`

type `M29`
expected answer is `Done saving file.`

DenisG1994, 2020-04-19:
All was smooth
```
M29
M29
ok
ok
M28 /mycode2.gco
M28 /mycode2.gco
echo:Now fresh file: /mycode2.gco
Writing to file: /mycode2.gco
ok
echo:Now fresh file: /mycode2.gco
Writing to file: /mycode2.gco
ok
M117 Hi man
M117 Hi man
ok
ok
G1 X20 F1000
G1 X20 F1000
ok
ok
M29
M29
Done saving file.
ok
Done saving file.
ok
```
MYCODE.GCO is now present on the SD card

luc-github, 2020-04-19:
there is no way currently to disable the check sum I need some time to see what to do
but at least it can be done

luc-github, 2020-04-19:
@DenisG1994 please try latest 2.1-devt actually  I have already prepared the possibility and it was just missing a flag to disable it

DenisG1994, 2020-04-19:
It was in fact the stock firmware... With the "custom" one it work:
```
N1 M29*25
ok
M21
echo:SD card ok
ok
M110 N0
ok
M117 Uploading...
ok
N1 M28 /test.gco*68
echo:Now fresh file: /test.gco
Writing to file: /test.gco
ok
N2 G28*17
ok
N3 M117 done*7
ok
N4 M29*28
Done saving file.
ok
M117 SD upload done
ok
Ok
M20
Begin file list
TEST.GCO 16
End file list
ok
```

DenisG1994, 2020-04-19:
give me 5 min, I will refresh the "old" firmware and retry

DenisG1994, 2020-04-19:
Ok, It work also with the stock firmware now!
```
[ESP3D][syncwebserver.cpp:943] handle_web_command(): WebCommand [ESP400]
[ESP3D][syncwebserver.cpp:858] handle_not_found(): Not found /macrocfg.json, type application/octet-stream
[ESP3D][syncwebserver.cpp:891] handle_not_found(): Page not found
[ESP3D][syncwebserver.cpp:858] handle_not_found(): Not found /preferences.json, type application/octet-stream
[ESP3D][syncwebserver.cpp:1227] SDFile_serial_upload(): Upload start
[ESP3D][webinterface.cpp:156] sendLine2Serial(): Send line 1
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
M29
ok
[ESP3D][webinterface.cpp:244] sendLine2Serial(): Got ok
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
[ESP3D][webinterface.cpp:156] sendLine2Serial(): Send line 1
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
M21
echo:SD card ok
ok
[ESP3D][webinterface.cpp:244] sendLine2Serial(): Got ok
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
[ESP3D][webinterface.cpp:156] sendLine2Serial(): Send line 1
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
M110 N0
ok
[ESP3D][webinterface.cpp:244] sendLine2Serial(): Got ok
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
M117 Uploading...
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
ok
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
[ESP3D][webinterface.cpp:156] sendLine2Serial(): Send line 1
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
M28 /test.gco
echo:Now fresh file: /test.gco
Writing to file: /test.gco
ok
[ESP3D][webinterface.cpp:244] sendLine2Serial(): Got ok
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
[ESP3D][syncwebserver.cpp:1284] SDFile_serial_upload(): Creation Ok
[ESP3D][webinterface.cpp:156] sendLine2Serial(): Send line 1
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
G28
ok
[ESP3D][webinterface.cpp:244] sendLine2Serial(): Got ok
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
[ESP3D][webinterface.cpp:156] sendLine2Serial(): Send line 1
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
M117 done
ok
[ESP3D][webinterface.cpp:244] sendLine2Serial(): Got ok
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
[ESP3D][syncwebserver.cpp:1358] SDFile_serial_upload(): Upload finished
[ESP3D][webinterface.cpp:99] purge_serial(): Purge Serial
[ESP3D][webinterface.cpp:118] purge_serial(): Purge done
[... 39 lines cut ...]
TEST.GCO
[ESP3D][syncwebserver.cpp:1028] handle_web_command(): Remove new line
[ESP3D][syncwebserver.cpp:1067] handle_web_command(): New buffer Begin file list
BASE~1.GCO
TEST.GCO
End file list
[ESP3D][syncwebserver.cpp:1028] handle_web_command(): Remove new line
[ESP3D][syncwebserver.cpp:1037] handle_web_command(): Found ok/wait add New buffer Begin file list
BASE~1.GCO
TEST.GCO
End file list
ok
[ESP3D][syncwebserver.cpp:1090] handle_web_command(): Finished
[ESP3D][syncwebserver.cpp:1094] handle_web_command(): Sending Begin file list
BASE~1.GCO
TEST.GCO
End file list
ok
[ESP3D][syncwebserver.cpp:1101] handle_web_command(): Start PurgeSerial
[ESP3D][syncwebserver.cpp:1106] handle_web_command(): End PurgeSerial
[ESP3D][syncwebserver.cpp:1108] handle_web_command(): Release PurgeSerial
```
Unlikely the printer don't seem to like the ".GCO" files, and don't allow to print them via the on-machine display. Guess I will print via the ESP, so power loss recovery is away in all cases.

[image]

luc-github, 2020-04-19:
well marlin is displaying only the file it can print so if it show it should print

luc-github, 2020-04-19:
I guess issue can be closed

closed 2020-04-19 by luc-github

DenisG1994, 2020-04-19:
Do I need to keep using the 2.1-devt version with all of the command shown? or you will release a separated version?

luc-github, 2020-04-19:
comment the DEBUG_OUTPUT_SOCKET in config.h

luc-github, 2020-04-19:
or download the latest 2.1-devt and enable the define 
```//For FW which has issue with checksum or not handling M110 properly///////
//#define DISABLE_SERIAL_CHECKSUM```

DenisG1994, 2020-04-19:
Ok, thank you very much for your assistance!!