luc-github/ESP3D-WEBUI #97: Is this the right version I am using?

oldman4U, 2020-05-08:
Hi Luc.

First I would like to thank you for all the time you spend to make this happen. I saw your recent "BTT/TFT Screen" post and started to configure the two original BTT 1MB -S based WIFI modules which I had lying around for some time now. It seems that I got everything up and running, so I can control my printers, but I would like to ask you if the version numbers UI 2.1b39 and FW 2.1.0.c26 are the one I should use.

Merci beaucoup

Georg
 
[image]

luc-github, 2020-05-08:
the BTT support is not yet in MArlin as I did not get any feedback - and I do not want to trust on my only own testing 😸 

the webui for BTT is currently here : https://github.com/luc-github/ESP3D-WEBUI/tree/2.1-BTT-TFT
web_ui_version = "2.1b60"
the FW is the latest git : https://github.com/luc-github/ESP3D/tree/2.1
 FW_VERSION "2.1.1.b6"

oldman4U, 2020-05-08:
Thats great, thank you.

I use one WIFI module with a TFT35 E3 and one connected to a SKR Pro v1.1. Should i use the webUI for both?

Last question if OK for you. Do you know if there is a way to send a gcode file directly from Cura to the ESP?

oldman4U, 2020-05-08:
Btw, this is the Commit which inspired me to give it a second chance.
[image]

luc-github, 2020-05-08:
>I use one WIFI module with a TFT35 E3 and one connected to a SKR Pro v1.1. Should i use the webUI for both?

yes you can but only enable TFT settings in webui to the one which is connected to TFT - I think it make sense

>Do you know if there is a way to send a gcode file directly from Cura to the ESP

I do not use Cura sorry but I know there are lot of plugin for Cura so you may check
but I would not recommand to stream GCODE by wifi even it is possible the risk of failure/blob is pretty high if your network is not fast enough 

I do not have any SKR pro so never checked the performance so I guess it is like having any ESP connected to serial, but IMHO I do not think you need ESP board on both skr pro and tft,  
Also do not expect to have the esp connected to skr be able to communicate with TFT - I think it won't work 

I hope it help

oldman4U, 2020-05-08:
Sorry, my fault.

These are two different printers.

The Cura connection is more about being able to send the gcode file over WIFI to the SD card of the TFT or the SKR instead of using the web interface. Today the workflow is:
1. save in Cura to file
2. open web UI
3. select gcode file
4. upload to SDcard of TFT or SKR board

Cura allows Save to File and save to removable disk, once it recognises such a disk or print over USB. Here a 4th method would be great which allows to save the gcode file over network to the SD card.

Anyway, i got more than enough of your time, i appreciate your time.

I will test the versions above, let you know and close this ticket right after that;-)

Greetings

Georg

oldman4U, 2020-05-23:
Hi Luc.

Sorry for not coming back to you earlier. I haven't had time to do the next steps but hope to be able to do so tomorrow. I will let you know for sure and close the ticket once we do not need it anymore.

Thank you

oldman4U, 2020-05-24:
Hi.

So no luck;-(

I downloaded the versions of the two links you sent to me, compiled the firmware, but found no way at all to compile the WebUI. I am running VSC on macOS, not Windows.

After compiling I found the firmware.bin file under .pioenvs/esp32dev, renamed it too esp3d.bin and uploaded the firmware (which is more than 1MB) using the currently installed WEB UI to the WIFI board. After a restart, the firmware is unchanged.

It would be great, if you could point me into the right direction and SORRY, if I missed a documentation which describes what I am asking for. But I searched for a documentation, but I could not find it.

Thank you

Georg

luc-github, 2020-05-24:
1 - you do not need to compile webUI the index.html.gz is already on repository of webUI in multilanguage as well as single / dual language which make it smaller and better for esp01 flash size

2 - /esp32dev is for esp32 not esp8266, you need to use the esp8266 fw AND as it seems you use platformIO you need to change the partition size to fit the esp01 as by default it is forMB flash not 1MB flash

oldman4U, 2020-05-24:
Little steps forward.

Web UI is now UI: 2.1b62 which seems to be the last recent version - thank you.

Firmware. I read and understand that my ESP01S has only 1MB, but I have no idea how I can change the partition size of it. I uploaded no language kit actively, I am fine with English only. 
_Note for ESP8266 1MB flash : FW is now too big will all features you need to chose strip the FW and select only some features, also WebUI is now also too big for full multilanguage support to fit the 128K SPIFFS so please use pack with limited language (en +another) https://github.com/luc-github/ESP3D-WEBUI/tree/2.1/languages_

Would be great if you could let me know how to change the partition size.

Thank you

luc-github, 2020-05-24:
please read : https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size
eagle.flash.1m144.ld should be ok for english only webUI as it is less than 100KB

oldman4U, 2020-05-24:
Hi Luc.

I completely understand that there are several stages of knowledge as well as interest between you and users like me, when it comes to computer and programming. You can write code and I struggle to understand the manuals, because I do not know the background good enough. When I try to help other users I do so, because I speak the amateur language they hopefully understand.

Just for fun, I uploaded one language package to one of the WIFI modules I have, and it seems that this was successful, because when I try to connect now, all I get is a LOADING....

I understand that this module does not have enough memory to run the latest version but I do not have enough time to read all the documents needed to get this module working properly. Every time I believe to be close to the solution, another issue or thing which has to be done pops up. You know the story with the Donkey and the carrot - it is a little bit this feeling.

Your help was very professional and I understand that developers like you do not get paid directly for what you do. It is great that you spend so much time to develop this software and even more, to help users like me. Once I get the 4MB memory modules I ordered in the meantime and once I find some time to change the memory I will do so. Maybe I can upload your software then and enjoy my carrot finally;-)

Georg

luc-github, 2020-05-24:
actually 1MB can be flashed with latest version of ESP3D and now it can even suport web update using latest core https://github.com/luc-github/ESP3D/issues/331#issuecomment-626325831

oldman4U, 2020-05-25:
Hi Luc.

After uploading the language file, I can not access the WIFI module any more, so I have only one left which works with the old software version. I do not want to break the second unit also, so I will wait for the 4MB memory and hope I am able to upgrade at least the inaccessible unit.

Thank you

luc-github, 2020-05-25:
you may have corrupted the fs by overflow the  fs
you may try the fallback access to deleted index.html.gz  file and upload again

oldman4U, 2020-05-25:
How can I delete it without being able to access it through the web interface?

After the upload it showed me 98% used

luc-github, 2020-05-25:
as show in read me use the embedded page https://raw.githubusercontent.com/luc-github/ESP3D/2.1/images/docs/embedded.png
type http://yourip?forcefallback=yes

oldman4U, 2020-05-25:
This is cool. Very cool. Will try it asap and let you know.  Don’t know how
I could miss it!!??🤪

oldman4U, 2020-05-26:
Thank you Luc, my second WIFI unit is back and up and running. Attached the screenshots from before and after.
[image]
[image]

Softwareversion on both units is now:
UI: 2.1b62 / FW: 2.1.0.c26

I will try:
actually 1MB can be flashed with latest version of ESP3D and now it can even suport web update using latest core luc-github/ESP3D#331 (comment)
asap.

luc-github, 2020-05-26:
no no no ^_^
you must not use index.html, you must use index.html.gz
using such big FS you won't be able to do webUpdate
you should use a SPIFFS around 160K or 144KB when you flash FW

oldman4U, 2020-05-26:
The index.html was the one which did not work. index.html.gz with 132.86KB is the current version which allows me to connect.

Is there a way to make this also using VSC? and if so, how - please;-)

_remove printf float support - currently not used

go to your ESP8266 core directory to disable float support
type python ./tools/boards.txt.py --nofloat --allgen

be noted this may affect your others ESP sketchs if you disable float support
select in debug mode of IDE : NoAssert-NDEBUG_

luc-github, 2020-05-26:
seems possible but never did myself : https://github.com/luc-github/ESP3D/issues/446

Anyway I did recent test and you can avoid it using new core feature : https://github.com/luc-github/ESP3D/issues/331#issuecomment-626325831

oldman4U, 2020-05-26:
#331

[image]

luc-github, 2020-05-26:
https://github.com/luc-github/ESP3D/blob/2.1/esp3d/config.h

oldman4U, 2020-05-26:
Because the link only shows the config but I can not use it, and because /blob does not exist, I downloaded the v2.1-devt and copied the config.h from there to v2.1. This allowed me to compile which resulted in a file Checking size .pioenvs/esp8266/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  36.7% (used 30040 bytes from 81920 bytes)
Flash: [====      ]  37.0% (used 386708 bytes from 1044464 bytes)
Creating BIN file ".pioenvs/esp8266/firmware.bin" using ".pioenvs/esp8266/firmware.elf"

So I renamed the firmware to esp3d.bin and uploaded it to the device. After that I can not access the device any more. Reverting to the "original" firmware, brings the device back.

luc-github, 2020-05-26:
no, you need to use full code not only the config.h 
also what flash size for filesystem did you used did you used  ?

oldman4U, 2020-05-27:
I downloaded the v2.1-devt and copied the config.h from there to v2.1. This allowed me to compile which resulted in a file Checking size .pioenvs/esp8266/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM: [==== ] 36.7% (used 30040 bytes from 81920 bytes)
Flash: [==== ] 37.0% (used 386708 bytes from 1044464 bytes)
Creating BIN file ".pioenvs/esp8266/firmware.bin" using ".pioenvs/esp8266/firmware.elf"

So I renamed the firmware to esp3d.bin and uploaded it to the device. After that I can not access the device any more. Reverting to the "original" firmware, brings the device back.

I did not change any flash size. How to change it and what size should i use?

luc-github, 2020-05-27:
>I downloaded the v2.1-devt and copied the config.h from there to v2.1

Same comment I gave you as before : https://github.com/luc-github/ESP3D-WEBUI/issues/97#issuecomment-634249473
no, you need to use full code not only the config.h

>I did not change any flash size. 

I aleready explained you that you need to to it : https://github.com/luc-github/ESP3D-WEBUI/issues/97#issuecomment-633249512

>How to change it and what size should i use?

same question you asked  and I give same answer https://github.com/luc-github/ESP3D-WEBUI/issues/97#issuecomment-633272943

oldman4U, 2020-05-27:
Hi Luc.

Again thank you for your help, I really appreciate. Like my nickname it already says - I am too old for this stuff.

Once I have time and do not know what to do, I will try it again.

Thank you

closed 2020-05-27 by oldman4U