luc-github/ESP3D-WEBUI #98: [BUG]

Tygrys-1, 2020-05-09:
**Describe the bug**
After running the dashboard (index.html) in browser, on the Commands monitor I have constatntly adding sequence:

X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
FR:100%
echo:E0 Flow: 100%

(is this ok?)
 
I can use the interface, move printer head, send gcode etc. But after ticking "Auto check every" in temperature panel the command window stops updating and interface stops working. Clicking on any control does nothing. After reseting the printer things gets back to normal. There are empty temperature diagrams (horizontal lines), no temperatures drawn.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Dashboard/(run the UI)
2. Click on "Auto check every" in temperature panel
3. No response form the printer. Any further request to the printer causes no response. 
4. Switching the printer off-on restores the communication.

**Expected behavior**
Earlier on UI 2.1b20 it worked - showing the diagrams and drawing the temperatures. I even managed to have this behavior once on b60, but now reverted to b20 and still the same problem. 

**ESP3D Firmware:  (please complete the following information)**
- UI: 2.1b60 / FW: 2.1.0.c26, tried BTT version and 2.1
- Wifi mode: STA
- Any change done in FW - none

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version 2.0.5.3

**Board used (please complete the following information):**
 - MCU: ESP32,  Running ESP on BTT TFT35 v3.0, fw 2.6.1 (latest)
 - Name:. ESP32-01S from BTT
 - Flash size: No idea

**Browser used (please complete the following information):**
- Name: Chrome, Firefox
- Version  Latest 
- Operating system Windows 10 64 bit

**Additional context**
Tried FW update, but despite the Update 100% message I have still the same FW. 2.1.0.c26

luc-github, 2020-05-09:
the messages you see 
```
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
FR:100%
echo:E0 Flow: 100%
```
are from TFT not ESP3D

to get proper BTT TFT support you need specific webui still in test :
https://github.com/luc-github/ESP3D-WEBUI/issues/97#issuecomment-625761677

luc-github, 2020-05-09:
also you cannot use the webupdate on esp01-s unless FW size is less than 400KB

Tygrys-1, 2020-05-09:
> 
> 
> the messages you see
> 
> ```
> X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
> FR:100%
> echo:E0 Flow: 100%
> ```
> 
> are from TFT not ESP3D
> 
> to get proper BTT TFT support you need specific webui still in test :
> [#97 (comment)](https://github.com/luc-github/ESP3D-WEBUI/issues/97#issuecomment-625761677)

I have no problem with this info flow, but the fact that activating thermal monitoring freezes the ESP.
I think that maybe the web UI waits for temp data, but receives this stream and stays in the waiting state forever.

Tygrys-1, 2020-05-09:
> also you cannot use the webupdate on esp01-s unless FW size is less than 400KB

Do you have any of that size? The last compiled is 500KB.

luc-github, 2020-05-09:
you need latest git of BTT PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/648 was merged 2 days ago so not yet in any release 

To decrease the FW size please check
https://github.com/luc-github/ESP3D/issues/331

But I am afraid you need to flash ESP manually

Tygrys-1, 2020-05-09:
For BTT: I downloaded from the Github yesterday from 2.6.1 branch. What is 2.5.4? Or should I rebuild it myself from master?

What is the flash size for BTT ESP32-01S, 512KB?

luc-github, 2020-05-09:
as wrote in comment: 
the webui for BTT is currently here : https://github.com/luc-github/ESP3D-WEBUI/tree/2.1-BTT-TFT
**web_ui_version = "2.1b60"**
the FW is the latest git : https://github.com/luc-github/ESP3D/tree/2.1
**FW_VERSION "2.1.1.b6"**
For TFT FW it is in master:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

you need to build esp3d by yourself, as well as BTT FW

ESP3D01-S Flash size is 1MB

Tygrys-1, 2020-05-09:
Thank you very much from help and sorry for the noob qestions. I'm quite new to this stuff.
And BTW, thank you for a great idea and this software.

Tygrys-1, 2020-05-09:
After upgrading Marlin to the latest bugfix and BTT TFT to the latest version and using the BTT version of the ESP3D got it working as it should. I'm still getting data flow in terminal window, it sometimes has a temp values, sometimes not, seems that switching verbose mode off can stop this flow from showing. Somehow ticking "Auto check every" affects the temperature display in the data flow, and in some cases stops the flow.

closed 2020-05-09 by Tygrys-1

luc-github, 2020-05-09:
the ESP3D `BTT version` is pretty old and I do not know if they do some change in my code, but latest esp3d has several bug fixes including some for BTT TFT

Tygrys-1, 2020-05-09:
The last UI 2.1 non BTT works the same as the BTT with the data flow. And it does not have TFT card / Udisk support. I do not see other difference.

luc-github, 2020-05-09:
if you do not see the settings of TFT SD / USB in WebUi  it means you do not use the one I linked to you.
in all case all work well with the versions I gave above