luc-github/ESP3D-WEBUI #99: ESP3D with MKS V1.4 smoothiebord

sns5400, 2020-05-18:
What is your question ?
Hello, i connect a MKS V1.4 smoothiebord with a ESP-Wroom-32 module and the ESP3D webui firmware on it. I can send commands on the webpage of the ESP3D like left movement, home, right movement etc but receive nothing from the smoothiebord, I using  P0.2 as RX and P0.3 as TX of the aux connector of the smoothiebord  to the ESP-wroom-32 TXD and RXD pin ,  speed in smoothiebord configfile is 115200. In the settings of the ESP3D i do not receive data from the smoothiebord, Have you any tip how to check the RX dataline ?

luc-github, 2020-05-18:
to check rx/tx lines you can use Serial 2 usb adaptor, connect it to your mks on rx/tx/ gnd and connect it to your pc.
open arduino terminal, setting baud rate properly and end lines as nl cf
send some command like M105 or ls /sd

luc-github, 2020-05-18:
of course be sure you enabled port in config.txt of smoothieware

sns5400, 2020-05-22:
Hello, you give me a good tip
I connect a usb2serial FTDI232 bord and open my arduino monitor, set speed at 115200.
When i send a command like M105 i see nothing, i must add the cariage return setting below the screen. Now i see a normal communication in the monitor. With M105 like you suggest i receive temp information. With the ls /sd i get the filename content of the sd card on the MKS bord.
What now ?

luc-github, 2020-05-22:
it means your serial port on printer board is working 
so now issue is between this port and esp board

sns5400, 2020-05-22:
On the pin-out of the esp-wroom-32 i find two tx/rx connections. Witch GPIO must i use to have correct communication

luc-github, 2020-05-22:
it depend how you configure esp3d, use the serial which you configured - by default is rx/tx

sns5400, 2020-05-23:
i configure nothing, i use de code direct from github  where can i configure this ? is there a special file or pin assosiation ?

luc-github, 2020-05-23:
it should work out of thebox if you use rx/tx,  config.h is the confugration file

sns5400, 2020-05-23:
Hello,
i using GPIO1 as TX and GPIO3 as RX So i connect the AUX connector MKS bord P0.2 --> GPIO3 (RX) and AUX connector P0.3 --> GPIO1 With this connection. When  i use the home buttons my printer respond. When i try to configure my printer i receive no variables or settings Receive data is my problem. Must i use a login to set the printer configuration ?

luc-github, 2020-05-23:
no need to login, if login is configured it will ask for it
I suggest you to open issue filling template - this is no more a 'question", the  issue template will ask for all informations for proper support as I have no idea of  the context

closed 2020-05-23 by luc-github

sns5400, 2020-05-24:
Found problem !  My module  rx0 pin do not work. I have change the esp32 for a older esp 8266 module and this function out of the box almost. When reading the printer config file, strange caracters in result (?) (?)(?)(?)(?)(?)(?)(?) is bizar.