luc-github/ESP3D-WEBUI #100: [FEATURE REQUEST] Allow more than 9 macros, more button colors

bryan065, 2020-06-12:
**Is your feature request related to a problem? Please describe.**
Allow more than 9 macros and more color options for the buttons (like green and light green). I'm adding custom gcode commands to control my printer and am running out of available macro slots. At the very least allow up to 15 or so macros?

**Describe the solution you'd like**
When the 9th macro is added, have a new blank macro line and allow something like up to 99 macros? Maybe allow the panel to scroll to view more (might have separate it from the controls panel).

luc-github, 2020-06-12:
it won't be possible in current webui
but may be in 3.0 it should be possible to increase macro numbers and color bg with custom colors

luc-github, 2020-10-02:
Still working on Macros but FR is filled - not Marco number limit - no color limit for text and background, but limit number of icons to ~100 to limit final package size
[image]

[image]

luc-github, 2020-10-03:
now in 3.0 webUI

closed 2020-10-03 by luc-github