luc-github/ESP3D-WEBUI #102: [Question] Hoe to access the SPIFFS : I've copy the index.html.gz from the filelist:Now Github there :(

Wesie, 2020-06-15:
I've a brandnew ESP3D, resolder the pins, so that I can connect the modul on the TFT (BTT) direct.
After call the <IP> I received the message, that the index.html.gz is missing, with a link to this side.... The Modul is still in Station Mode. 
After readinf a little bit ( I'll believe not to much :( ) I've download this file:
https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz
and copy the file on the Modul.
Now I receive at every time the copy from the Github, if I call the <IP> from the modul :( 
------------------------------
Skip to content
luc-github / ESP3D-WEBUI
Sign up

    Why GitHub?
                          
------------------------------          
All readings in the Internet doesn't help: Information are to old, sides no more available .... 
How I get an access to the SPIFFS memory to delete the file or how I can make a factory reset ?
Thx in adavanced
DIrk

luc-github, 2020-06-15:
you can access to embedded page by typing `http://<your IP>?forcefallback=yes `

[image]

Wesie, 2020-06-15:
YES, it's works :)
I've try a time ago, but I've forget (not recognise) the "?" .
Which file is now the correct one for  upload? In the right upper  corner V1.2 is written.
For the flash filesystem I get following information; Total space: 471.36 KB  |  Used space: 251 B

luc-github, 2020-06-15:
this one https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz?raw=true

Wesie, 2020-06-15:
Many Thank's . It's work now :)
About the memory I received this information:
Prozessor-ID:  13349399
Prozessorfrequenz:  160Mhz
Freier Speicher:  38.34 KB
SDK:  2.2.1(cfd48f3)
Flashgröße:  1.00 MB
Verfrügbare Grröße frür das Update:  131.59 KB(Not enough)
Verfrügbare Grröße frür das SPIFFS:  471.36 KB

I've no idea, for what the rest of the memory is reserviert or who used this one ....
Some Misspellings are incl:
Verfrügbare  >> Verfügbare 
Grröße >> Größe
frür >> für 
it looks likes, as is in front of every "Umlaut" a prefix "r" available.

luc-github, 2020-06-15:
the german come from contributor
you can review it and fix glitches using https://raw.githubusercontent.com/luc-github/ESP3D-WEBUI/2.1/www/js/language/de.js

Wesie, 2020-06-16:
I'll check the translation in the evening after work.
About the memory I've found 1 interssting thing. The memory flash is onboard:
From the datasheet it's a 1M chip:
`The EN25Q80B is an 8 Megabit (1024K-byte) Serial Flash memory, with advanced ...__ `
But in the datasheet of the :
ESP8266EX uses external SPI flash to store user programs, and supports up to 16 MB
memory capacity theoretically.
The minimum flash memory of ESP8266EX is shown below:
• OTA disabled: 512 kB at least
• OTA enabled: 1 MB at least__

Could it be possible, that in the firmware OTA is enabled, so that a part of the memory is reserved for an OTA update?

luc-github, 2020-06-16:
yes if you use the one for BTT it is an ESP01 with 1MB flash
but using so much for FS will definitly block you for OTA - and if you want to use future version of ESP3D it won't fit 
I suggest you to use 144KB for FS
OTA will work under 2.1 under some conditions (https://github.com/luc-github/ESP3D/issues/331#issuecomment-626325831) but on ESP connected to BTT board it is not a problem to update FW by putting binary on SD card from what I know

also if you want to control your BTT TFT I did a test webUI to control TFT SD and TFT USB and adding other improvements 
https://github.com/luc-github/ESP3D-WEBUI/tree/2.1-BTT-TFT/languages/de

Wesie, 2020-06-16:
> OTA will work under 2.1 under some conditions (luc-github/ESP3D#331
> (comment)

Hi,
please find enclosed the German translation without some error, which I've
found ;)

Best Regards
Dirk

//removeIf(de_lang_disabled)
//german
var germantrans = {
"de":"Deutsch",
"ESP3D for":"ESP3D f&uuml;r",
"Value of auto-check must be between 0s and 99s !!":"Der Wert des Auto-Checks muss zwischen 0s und 99s liegen !!",
"Value of extruder velocity must be between 1 mm/min and 9999 mm/min !":"Der Wert der Extrudergeschwindigkeit muss zwischen 1 mm / min und 9999 mm / min liegen !",
"Value of filament length must be between 0.001 mm and 9999 mm !":"Der wert der filament l&auml;nge muss zwischen 0.001mm und 9999mm liegen !",
"cannot have '-', '#' char or be empty":"'-', '#' d&uuml;fen nicht benutzt werden oder leer sein",
"cannot have '-', 'e' char or be empty":"'-', 'e' d&uuml;fen nicht benutzt werden oder leer sein",
"Failed:":"Gescheitert:",
"File config / config.txt not found!":"Datei config / config.txt nicht gefunden!",
"File name cannot be empty!":"Dateiname kann nicht leer sein!",
"Value must be ":"Wert muss sein ",
"Value must be between 0 degres and 999 degres !":"Wert muss zwischen 0 Grad und 999 Grad liegen !",
"Value must be between 0% and 100% !":"Wert muss zwischen 0% und 100% liegen !",
"Value must be between 25% and 150% !":"Wert muss zwischen 25% und 150% liegen !",
"Value must be between 50% and 300% !":"Wert muss zwischen 50% und 300% liegen !",
"XY feedrate value must be between 1 mm/min and 9999 mm/min !":"XY Vorschubgeschwindigkeit muss zwischen 1mm/min und 9999 mm/min liegen !",
"Z feedrate value must be between 1 mm/min and 999 mm/min !":"Z Vorschubgeschwindigkeit muss zwischen 1mm/min und 999 mm/min liegen !",
" seconds":" Sekunden",
"Abort":"Abbruch",
"auto-check every:":"Auto Check alle:",
"auto-check position every:":"Auto Check der Position alle:",
"Autoscroll":"Autoscrollen",
"Bed":"Heizbett",
"Board":"Board",
"Busy...":"Ausgelastet...",
"Camera":"Kamera",
"Cancel":"Abbrechen",
"Cannot get EEPROM content!":"Kann EEPROM Inhalt nicht auslesen",
"Clear":"L&ouml;schen",
"Close":"Schliessen",
"Color":"Farbe",
"Commands":"Kommandos",
"Communication locked by another process, retry later.":"Kommunikation ist gesperrt von einem anderen Prozess, sp&auml;ter nochmal probieren.",
"Communication locked!":"Kommunikation gesperrt!",
"Communications are currently locked, please wait and retry.":"Kommunikation vorr&uuml;bergehen gesperrt, bitte warten und wiederholen!",
"Confirm deletion of directory: ":"Best&auml;tigen um das Verzeichnis zu l&ouml;schen: ",
"Confirm deletion of file: ":"Best&auml;tigen um die Datei zu l&ouml;schen: ",
"Connecting ESP3D...":"Verbindungsaufbau ESP3D...",
"Connection failed! is your FW correct?":"Verbindung gescheitert! Richtige Firmware ausgew&auml;hlt?",
"Controls":"Steuerung",
"Credits":"ZU Verdanken",
"Dashboard":"&Uuml;bersichtsseite",
"Data mofified":"Daten ge&auml;ndert",
"Do you want to save?":"Sollen gespeichert werden?",
"Enable second extruder controls":"Zweiten extruder aktivieren",
"Error":"Fehler",
"ESP3D Filesystem":"ESP3D Dateisystem",
"ESP3D Settings":"ESP3D Einstellungen",
"ESP3D Status":"ESP3D Status",
"ESP3D Update":"ESP3D Aktualisierung",
"Extrude":"Extruieren",
"Extruder T0":"Extruder T0",
"Extruder T1":"Extruder T1",
"Extruders":"Extruder",
"Fan (0-100%)":"L&uuml;fter (0-100%)",
"Feed (25-150%)":"Vorschub (25-150%)",
"Feedrate :":"Vorschubgeschwindigkeit :",
"Filename":"Dateiname",
"Filename/URI":"Dateiname/URI",
"Verbose mode":"Ausf&uuml;hrlicher modus",
[... 281 lines cut ...]
"error:22":"Feed rate has not yet been set or is undefined.",
"error:23":"G-code command in block requires an integer value.",
"error:24":"More than one g-code command that requires axis words found in block.",
"error:25":"Repeated g-code word found in block.",
"error:26":"No axis words found in block for g-code command or current modal state which requires them.",
"error:27":"Line number value is invalid.",
"error:28":"G-code command is missing a required value word.",
"error:29":"G59.x work coordinate systems are not supported.",
"error:30":"G53 only allowed with G0 and G1 motion modes.",
"error:31":"Axis words found in block when no command or current modal state uses them.",
"error:32":"G2 and G3 arcs require at least one in-plane axis word.",
"error:33":"Motion command target is invalid.",
"error:34":"Arc radius value is invalid.",
"error:35":"G2 and G3 arcs require at least one in-plane offset word.",
"error:36":"Unused value words found in block.",
"error:37":"G43.1 dynamic tool length offset is not assigned to configured tool length axis.",
"error:38":"Tool number greater than max supported value.",
"error:60":"SD failed to mount",
"error:61":"SD card failed to open file for reading",
"error:62":"SD card failed to open directory",
"error:63":"SD Card directory not found",
"error:64":"SD Card file empty",
"error:70":"Bluetooth failed to start",
};
//endRemoveIf(de_lang_disabled)

luc-github, 2020-06-16:
Updated - thank you ^_^

Wesie, 2020-06-16:
Update No.2
Only the Alarms and Error msg are still in English ....
I've search a little bit in the GUI to lock what it's means ;)

Best regards
Dirk

//removeIf(de_lang_disabled)
//german
var germantrans = {
"de":"Deutsch",
"ESP3D for":"ESP3D f&uuml;r",
"Value of auto-check must be between 0s and 99s !!":"Der Wert des Auto-Checks muss zwischen 0s und 99s liegen !!",
"Value of extruder velocity must be between 1 mm/min and 9999 mm/min !":"Der Wert der Extrudergeschwindigkeit muss zwischen 1 mm/min und 9999 mm/min liegen !",
"Value of filament length must be between 0.001 mm and 9999 mm !":"Der Wert der Filamentl&auml;nge muss zwischen 0.001mm und 9999mm liegen !",
"cannot have '-', '#' char or be empty":"'-', '#' d&uuml;rfen nicht benutzt werden oder leer sein",
"cannot have '-', 'e' char or be empty":"'-', 'e' d&uuml;rfen nicht benutzt werden oder leer sein",
"Failed:":"Gescheitert:",
"File config / config.txt not found!":"Datei config / config.txt nicht gefunden!",
"File name cannot be empty!":"Dateiname kann nicht leer sein!",
"Value must be ":"Der Wert muss sein ",
"Value must be between 0 degres and 999 degres !":"Der Wert muss zwischen 0 Grad und 999 Grad liegen !",
"Value must be between 0% and 100% !":"Der Wert muss zwischen 0% und 100% liegen !",
"Value must be between 25% and 150% !":"Der Wert muss zwischen 25% und 150% liegen !",
"Value must be between 50% and 300% !":"Der Wert muss zwischen 50% und 300% liegen !",
"XY feedrate value must be between 1 mm/min and 9999 mm/min !":"XY Vorschubgeschwindigkeit muss zwischen 1mm/min und 9999 mm/min liegen !",
"Z feedrate value must be between 1 mm/min and 999 mm/min !":"Z Vorschubgeschwindigkeit muss zwischen 1mm/min und 999 mm/min liegen !",
" seconds":" Sekunden",
"Abort":"Abbruch",
"auto-check every:":"Auto Check alle:",
"auto-check position every:":"Auto Check jede Position alle:",
"Autoscroll":"Autoscrollen",
"Bed":"Heizbett",
"Board":"Board",
"Busy...":"Ausgelastet...",
"Camera":"Kamera",
"Cancel":"Abbrechen",
"Cannot get EEPROM content!":"Kann EEPROM Inhalt nicht auslesen",
"Clear":"L&ouml;schen",
"Close":"Schliessen",
"Color":"Farbe",
"Commands":"Kommandos",
"Communication locked by another process, retry later.":"Kommunikation ist gesperrt von einem anderen Prozess, sp&auml;ter nochmal probieren.",
"Communication locked!":"Kommunikation gesperrt!",
"Communications are currently locked, please wait and retry.":"Kommunikation vorr&uuml;bergehen gesperrt, bitte warten und wiederholen!",
"Confirm deletion of directory: ":"Best&auml;tigen um das Verzeichnis zu l&ouml;schen: ",
"Confirm deletion of file: ":"Best&auml;tigen um die Datei zu l&ouml;schen: ",
"Connecting ESP3D...":"Verbindungsaufbau ESP3D...",
"Connection failed! is your FW correct?":"Verbindung gescheitert! Richtige Firmware ausgew&auml;hlt?",
"Controls":"Steuerung",
"Credits":"Bedanken",
"Dashboard":"&Uuml;bersichtsseite",
"Data mofified":"Daten ge&auml;ndert",
"Do you want to save?":"Sollen gespeichert werden?",
"Enable second extruder controls":"Zweiten Extruder aktivieren",
"Error":"Fehler",
"ESP3D Filesystem":"ESP3D Dateisystem",
"ESP3D Settings":"ESP3D Einstellungen",
"ESP3D Status":"ESP3D Status",
"ESP3D Update":"ESP3D Aktualisierung",
"Extrude":"Extruieren",
"Extruder T0":"Extruder T0",
"Extruder T1":"Extruder T1",
"Extruders":"Extruder",
"Fan (0-100%)":"L&uuml;fter (0-100%)",
"Feed (25-150%)":"Vorschub (25-150%)",
"Feedrate :":"Vorschubgeschwindigkeit :",
"Filename":"Dateiname",
"Filename/URI":"Dateiname/URI",
"Verbose mode":"Ausf&uuml;hrlicher Modus",
[... 281 lines cut ...]
"error:22":"Feed rate has not yet been set or is undefined.",
"error:23":"G-code command in block requires an integer value.",
"error:24":"More than one g-code command that requires axis words found in block.",
"error:25":"Repeated g-code word found in block.",
"error:26":"No axis words found in block for g-code command or current modal state which requires them.",
"error:27":"Line number value is invalid.",
"error:28":"G-code command is missing a required value word.",
"error:29":"G59.x work coordinate systems are not supported.",
"error:30":"G53 only allowed with G0 and G1 motion modes.",
"error:31":"Axis words found in block when no command or current modal state uses them.",
"error:32":"G2 and G3 arcs require at least one in-plane axis word.",
"error:33":"Motion command target is invalid.",
"error:34":"Arc radius value is invalid.",
"error:35":"G2 and G3 arcs require at least one in-plane offset word.",
"error:36":"Unused value words found in block.",
"error:37":"G43.1 dynamic tool length offset is not assigned to configured tool length axis.",
"error:38":"Tool number greater than max supported value.",
"error:60":"SD failed to mount",
"error:61":"SD card failed to open file for reading",
"error:62":"SD card failed to open directory",
"error:63":"SD Card directory not found",
"error:64":"SD Card file empty",
"error:70":"Bluetooth failed to start",
};
//endRemoveIf(de_lang_disabled)

luc-github, 2020-06-16:
Alarms and Error msg are for grbl not 3d printing so if you do not use grbl you won't see them

Wesie, 2020-06-17:
Good morning,
thanks for the link for GRBL.
One question : Why I can found the needed image (gz) by myself? I can only
dl the files with your links. If I'm search on the HP of Github, I couldn't
find the file for the Web-Gui, nor for the Firmware.. only the sources.
Where is my mistake?
First look on the Gui for GRBL and the tft35 BTT look good. I can read read
the folder of the SD Card and start a file. The USB port make some
trouble.. If I can read the USB on the Display by itself, I couldn't
received an answer out from the the Web-Gui. Maybe a timing issue?
I must investigate it a little bit deeper, because I changed the firmware
in this moment for a mini milling machine CNC 3018 pro (away from a
printer) ; But I use the global and needed routine like the ring buffer and
so on ;)

Best Regards
Dirk

luc-github, 2020-06-17:
all index.htlm.gz are here : 
https://github.com/luc-github/ESP3D-WEBUI
[image]
they are available for each language and for multi also

you use GRBL FW on your CNC3018 pro with BTT TFT and it works ? I am surprised I though it was only working with Marlin 

For your cnc I suggest to use this project : https://github.com/bdring/Grbl_Esp32 it use same webUI but using directly and ESP32 board instead of daughter board which make the job from SD safer and faster than host streaming

Wesie, 2020-06-17:
Hi,
so as mention before: I modify the firmware from the TFT35 from Marlin (3D
pronter) to Grbl (CNC). Most things are working now.
In this moment a few things are still missing : Preselect Spindel speed and
Feed-rate, Z-Probe and XYZ Probe. Manual Menue to move all axis with
spindle on, off .... organise the menus a little bit better, remove the
uneedet code. But it's going forward :)
Best Regards
Dirk

luc-github, 2020-06-17:
> I modify the firmware from the TFT35 from Marlin (3D
pronter) to Grbl (CNC).

Ho I misunderstood this part sorry

Wesie, 2020-06-19:
Good morning,
 till this moment I haven't found any time to investigate deeper into the
webgui, because I must modify and create more menu pages and create own
Icons for the normal usage, so I must try, what's working or not (e. g.
presetting  Feed rate and presetting Spindel rate in front (sending via G21
F500)  will not be reflect back in the Status message via "?".
But on the weekend I've the item Webgui on my pers. plan. So I expect, that
I can send some questions, results in the next few days to you :)

Best Regards
Dirk

luc-github, 2020-06-19:
I am rewriting webUI for esp3D 3.0 - the GRBL interface should be improved because 2.1 it is basic and more derivated from 3d printer one - in 3.0 it should be fully cnc because now code are splitted - let me know what it is currently missing in 2.1 so I may work on it

Wesie, 2020-06-19:
Hi Luc,
the issue with reading the USB looks independent of the seize of the
USB-Stick. With a small USB reading via WEB is OK, today. .... Further
investigation is needed ..
Another thing:
In the Settings I can say, Show me the TFT-SD / TF-USB. Uncomfortable Grbl
has no internal SD.

Wish:
a) Possibility to disable the normal SD in the GUI (not visible)
b) If the surface is secured (upper right tag) its not possible to change
between the different file sources (TFT SD to TFT-USB and vice versa) . It
will be nice, if this option is from lock excluded:
[image: grafik.png]

luc-github, 2020-06-19:
>In the Settings I can say, Show me the TFT-SD / TF-USB. Uncomfortable Grbl
has no internal SD.

GRBL_ESP32 support SD - in 2.1 TFT support was  added and not in idea of having TFT-SD without SD on board as TFT was not supporting GRBL so the solution was to remove the File interface - your case GRBL no SD  - BTT TFT SD for GRBL was not in expected case
In 3.0 it will be ok because the selection is more flexible in settings 

>If the surface is secured (upper right tag) its not possible to change
between the different file sources (TFT SD to TFT-USB and vice versa) .

yes UI is locked - because it query ESP which query TFT or Printer/CNC,
In 3.0 I did not implemented the lock interface yet - and there is more file cache implemented - so I think changing selection without query the TFT when interface is locked should be possible - that said, if no previous query was done when interface is unlock - changing selection  when UI locked will display nothing

luc-github, 2020-06-19:
@Wesie do you have some code I could test for your BTT GRBL TFT?

Wesie, 2020-06-20:
Does you have a BTT Display and GRBL as controller?

luc-github, 2020-06-20:
yes I do

Wesie, 2020-06-20:
Oh, that's great, because I've build the CNC3-3018 pro and after add
optical Endstops I've recognize that the default small offline controller
could not support Homing. So this one is useless for me. Than I've start
with the Display BTT 35V3 (with knob). Which display does you have, because
the code must compile for the right hardware typ. Hopefully not a small
one, because the additional icons have the seize for the 3,5 Display. Maybe
the 3.0 Display use the same seize. All other Displays need smaller icons,
which I haven't create till this moment....
Dirk

luc-github, 2020-06-20:
I have BTT TFT 35 - E3 V3.0 / BTT TFT 3.5 V3.0
I have also flashed My MKS 32 V4.0 and V2.0 with BTT FW

Wesie, 2020-06-20:
That's great, I'll send you the complied bin and the icons in the next
time,. I must write also some hints, what is add changed and added :)

Wesie, 2020-06-20:
Hi Luc,

I must prepare myself to walk with our dog.
Pls find enclosed the bin and the icon to flash on the BTT V3 3.5
The connection is only to the CNC with the serial port beside the 3 block
connector. More later..

luc-github, 2020-06-20:
@Wesie you need to use gtihub to attach files
everytime you attach files by email they are not displayed / available

Wesie, 2020-06-20:
hi Luc,
on Github I can only make an attachemnt via link. 
The code is still in progress .... so nothing to publish now in this moment. Can you send me your  mail address, so I can send the files direct to you?

luc-github, 2020-06-20:
you just drop a zip file [ESP3D-features.zip](https://github.com/luc-github/ESP3D-WEBUI/files/4807684/ESP3D-features.zip)
like I did

Wesie, 2020-06-20:
[BIGTREE_TFT35_V3.0.26.1.zip](https://github.com/luc-github/ESP3D-WEBUI/files/4807730/BIGTREE_TFT35_V3.0.26.1.zip)

[config.zip](https://github.com/luc-github/ESP3D-WEBUI/files/4807732/config.zip)

In the config.ini the Serial port must be always on, (compare to my attached config.ini) and the min/max Values for XYZ should matched for your device. (without drive into the endstops.)
0.0.0 is on my device on the right top corner and Z the highes position. All ways runs into the zero area ...on my side.... like -100 (X),-50(Y),-5(Z).

The first Alarm or Error. will stay in the notification field and must be confirmed.
All other Alams and Error are ignored till confirmation. 
Other  the info field: here the last messsge is visible. Delete the msg could be done via one-Touch on the infofield.

The status report in Grbl should be "$10=3" . Grbl should be version 1.1 

In Movements, there is the possibility to send $X and $H, also Jogging to different places (in depent of the min, max values of your config). In this moment , you can set the Speed and feed for each axis, which is stored localy on the TFT during runtime. (Statusscreen right upper icon = one touch to go in)
This is in preparation if the manual movement menue will be available. 
The probe ( Z or for all 3 axis is still missing, a lot of "old" code for a 3D pronter is still avaialble ..... It's only a snapshoot ;)

luc-github, 2020-06-20:
@Wesie thank you - I will have a check

Wesie, 2020-06-20:
Could you get a connection? Jooging? Homing,....?

luc-github, 2020-06-21:
did not tested yet - will feedback when done

luc-github, 2020-07-12:
I do not have available system for GRBL for test right now as I am currently on the 3.0 and focusing on printer part - I will report later - so for now I think this issue can be closed

closed 2020-07-12 by luc-github