luc-github/ESP3D-WEBUI #104: Last release does not work with ESP firmware

leseaw, 2020-07-12:
What is your question ?

Can't you please make a new release for the ESP firmware?
Because the stand from here and from there do not work on a NodeMCU 12E
With the latest firmware version it works immediately.

thanks

luc-github, 2020-07-12:
what do you mean ? what does not work ?

leseaw, 2020-07-12:
download last release from the firmware
Upload
ESP is running, but on startup it wants to have index.gz file
Upload this file from the WebGui release
ESP no longer starts
Only by uploading the firmware again it will come back
But with the latest self-built firmware it works fine

therefore
we should have a new release of the firmware
so that no one else has a problem with it

luc-github, 2020-07-12:
please give versions (ESP3D and WEBUI)  you use and meet issue and the ones you do not as I cannot reproduce 
also FW release version go with WEBUI release version
FW git version go with WEBUI git version

luc-github, 2020-07-12:
I have just tested 
https://github.com/luc-github/ESP3D-WEBUI/releases/tag/v2.1
https://github.com/luc-github/ESP3D/releases/tag/2.1

and no issue
same for GIT versions

leseaw, 2020-07-12:
will check tomorrow (Modellnumber from ESP Modul)
Build with Arduibo Ide

leseaw, 2020-07-15:
Strange,

other identical NodeMCU does not make these trouble.
With the other board I can always readjust it.
This requires newer firmware that it works.
Actually both are identical.
Memory is also completely erased before writing.

luc-github, 2020-07-15:
be sure you did not changed partition size because this may corrupt the filesystem

closed 2020-07-19 by luc-github