luc-github/ESP3D-WEBUI #106: [FEATURE REQUEST] Bed Mesh Leveling Visualizer

dmiart, 2020-07-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I am using an octoprint plugin called bed visualizer to display G29 T. Would be nice to have something like this in the esp3D. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I basically want to be able to see this a graph like this on the web page of the ESP3D. Or at least a very basic graph that shows 4 corners and how much they differ from each other, so I know which knobs to turn. 

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

luc-github, 2020-07-23:
can you share a G29 T output ? I do not have probe for bed leveling

dmiart, 2020-07-23:
> 
> 
> can you share a G29 T output ? I do not have probe for bed leveling

//action:notification Ender3 skr 1.3 2/08/20v2 Ready.
Bilinear Leveling Grid:
      0      1      2
 0 +0.172 +0.085 -0.058
 1 +0.189 +0.037 -0.131
 2 +0.095 -0.084 -0.183

luc-github, 2020-07-24:
ok clear, thank you

luc-github, 2020-07-25:
I think to do this representation need user to key bed size or we can also use an `virtual` square just described by external probed points

dmiart, 2020-07-26:
Is there no way to get it during the runtime from the printer itself? What about moving the hotend x and y to the max possible and then getting that position ?

luc-github, 2020-07-27:
Sorry I do not understand
Do you mean you Use G29 V2 ... to get it verbose and get updated report when running ? Does it work ?

dmiart, 2020-08-12:
I am not entirely sure what you mean. That plugin does G29 T I think.  For the bed size, I meant wouldn't it be possible to query the printer for that information or to force it to move as far to the right and forward as possible to determine the bed size?

luc-github, 2020-08-13:
my answer was about :
>Is there no way to get it during the runtime from the printer itself?

what do you mean ?

For bed size there is no GCODE I am aware of to get it (I think only prusa FW has something in debug mode) so I think user user need to key info

dmiart, 2020-08-13:
> 
> 
> my answer was about :
> 
> > Is there no way to get it during the runtime from the printer itself?
> 
> what do you mean ?
> 
> For bed size there is no GCODE I am aware of to get it (I think only prusa FW has something in debug mode) so I think user user need to key info

Strange that there is no M or G code for it. On second though letting user enter it would probably not be too bad since it only has to be done once.

luc-github, 2020-08-13:
Well there is one command in Marlin that may be used but it is not the bed size , it is endstop positions - which may differe from bed size and also need some define set in Marlin FW to work
```
M211
echo:Soft endstops: ON  Min:  X0.00 Y0.00 Z0.00   Max:  X220.00 Y240.00 Z130.00
```

Cannot find anyother GCODE in Repetier / Smoothieware neither, if you can find it let me know

luc-github, 2020-09-07:
as mentioned above so far there is no `print size`  but max position which even offset is defined may not give the correct bed size - 
max position is not always upper limit of bed size 

Also I am definitly  not Marlin focused - I prefer to find solution that benefit the most 

Thank you for the link I will have a look

I do not have delta printer neither auto level probe system 
what the difference of bed between delta or Cartesian printer for such purpose ?

Note: I know plotly but its footprint is too big and cannot be used, gzip version of mininimal code is around 340K which is too big

ETE-Design, 2020-11-01:
@luc-github  This would be a great feature if possible :-)

luc-github, 2020-11-02:
it is set as planned, so it is in TODO list but .. not on top items, also I do not have any bltouch / probe system, so it will need to wait until I get one to work on this,