luc-github/ESP3D-WEBUI #107: [BUG] Unable to connect after loading incorrect "index.tar.gz"

marciot, 2020-07-29:
**Describe the bug**
I upgraded ESP3D to version 3.0, but accidentally uploaded the "index.tar.gz" for 2.1. Now when I try to connect, I get a dialog box saying:

"Connection failed! Is your FW correct?"

It seems like this dialog box should have an option to re-upload the "index.tar.gz" in the case in which it gets corrupted.

I tried to reflash the FW, hoping that it would blow away the file system, but this seems to not have happened. Until I can find a way to erase the filesystem, I do not know how to fix this problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Install 3.0 ESP 3D
2. Install "index.tar.gz" for 2.1
3. Try to connect

**Expected behavior**
It should give you an option to re-upload the "index.html.gz" 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:  (please complete the following information)**
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]
- Wifi mode:[e.g. STA]
- Any change done in FW

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version 2.0

**Board used (please complete the following information):**
 - MCU: ESP8266 
 - Name: ESP-01S 
 - Flash size: 1MB

**Browser used (please complete the following information):**
- Name: Chrome
- Version  84.0.4147.105
- Operating system Windows 10

luc-github, 2020-07-30:
WebUI and FW need to match they are not compatible accross version 2.X vs 3.X
use the maintenance page : `http://<yourIP>?forcefallback=yes`

[image]

closed 2020-07-30 by luc-github

luc-github, 2020-07-30:
well it is documented  - it is in readme and the only screenshot of 2.1 : https://github.com/luc-github/ESP3D but no one read I know that....
so adding  such link may also be confusing to user as they won't understand the purpose of it ... because connection failed can be caused by other reason than invalid index.html.gz

I prefer to keep UI simple for the moment - but it may change in future