luc-github/ESP3D-WEBUI #108: Keyboard Shortcuts

BogdanTheGeek, 2020-08-07:
Hi, is there any way to setup keyboard shortcuts for jogging, overrides etc.?

luc-github, 2020-08-07:
not currently, is that possible in HTML ?

BogdanTheGeek, 2020-08-07:
Generally jogging is the most is the most important.
I think it can also be done easily with a keyboard event catcher in js and a switch case.
Is there any documentation for adding pluggins/ features?
I would like a gcode visualiser and editor in the ui. I think with these features it would getting almost to the level of good gcode senders.

BogdanTheGeek, 2020-08-07:
I should probably mention that I use Grbl-esp32 for a cnc, so these features are more targeted to that workflow than 3d printing. Is there any way to enable/ disable the steppers from the ui or with a macro?
PS: sorry for asking so many questions.

luc-github, 2020-08-07:
I will add this in 3.0
No plugin method yet sorry 

a gcode visualiser is challenging - webUI does not have access to SD content if no direct access to SD, same for editor and even it would have access, it would need to reload the whole GCODE each time the page is opened  - and I am not sure how to sync the rendering to the work in progress when opening page during job
webUI is not a GCODE sender I do not think it could compete easily with a PC host application 

any GCODE can be launched using macro file : https://github.com/bdring/Grbl_Esp32/wiki/ESP3D-Web-UI-for-Grbl_ESP32#macros

You can also create hardware buttons https://github.com/bdring/Grbl_Esp32/wiki/Macro-Button-Feature

luc-github, 2020-08-10:
Seems accessKey is not working in Preact when ok in pure HTML - not sure what is the issue, but will need some dig in

closed 2020-09-04 by stale[bot]