luc-github/ESP3D-WEBUI #110: Turkish Translation help for ESP3D WebUI [FEATURE REQUEST]

DusDus, 2020-08-20:
Hello,

I do not possess a great deal of knowledge about  coding; however, I wanted to have a Turkish WebUI so I tried to translate some part of the template. I tried to follow the guidelines you put out for translation and here is the file for Turkish translation:
[Trans-Turkish.txt](https://github.com/luc-github/ESP3D-WEBUI/files/5102060/Trans-Turkish.txt)

What else do I need to do?

Thank you for your time.

luc-github, 2020-08-20:
Hi @DusDus  thank you for your help, I need the missing entry
`"lang":"Lang in local language"`
it is `Turkish` in Turkish language

DusDus, 2020-08-20:
Hello,

Thank you for responding quickly. Here is the missing line:
"tr":"T&uuml;rk&ccedil;e"

Regards.

DusDus, 2020-08-20:
Hello,

Thank you for responding quickly. Here is the revised file:
[Trans-Turkish.txt](https://github.com/luc-github/ESP3D-WEBUI/files/5102295/Trans-Turkish.txt)

Regards.

luc-github, 2020-08-20:
I will addit in build script and git this week, thank you